me-no-dev/ESPAsyncWebServer #1309: Need help with simplifying of server requests hardcoding.

OleksandrGora, 2023-05-12:
Dear all.
I am new in programming. Try to develop server using this library, LittleFS, board ESP32. Faced with a necessity to hardcode server requests in case I change front end part of the project. In case I want to use additional icons I should hardcode requests for every single file. I tried to use loops through all files on the server.
This test code works (I was going to create an array with list all files and use a loop after that):
`const String fullPath[] = {"/style.css", "/code.js"};
server.on(fullPath[0].c_str(), HTTP_GET, [](AsyncWebServerRequest *request)
			  { request->send(LittleFS, fullPath[0]); }); 
server.on(fullPath[1].c_str(), HTTP_GET, [](AsyncWebServerRequest *request)
			  { request->send(LittleFS, fullPath[1]); }); `

But this one does not work:
`const String fullPath[] = {"/style.css", "/code.js"};
for (counter = 0; counter < 2; counter++)
	{
		server.on(fullPath[counter].c_str(), HTTP_GET, [](AsyncWebServerRequest *request)
				  { request->send(LittleFS, fullPath[counter]); });
	}`

Please point out on an error. Or please offer another way for solving the issue. Or please provide with a link where this (or similar) issue was resolved.

Thanks in advance.

sszczep, 2023-05-14:
What exactly seems to be an issue? Is it a compilation error or it doesn't work runtime?
You haven't declared `counter` variable but I guess you have it somewhere else in the code. 

Have you tried using `.serveStatic()` method?

OleksandrGora, 2023-05-14:
@sszczep: Thanks for .serveStatic() suggestion!

I tried and at the moment it seems it solved my issue!

closed 2023-05-14 by OleksandrGora