me-no-dev/ESPAsyncWebServer #1319: The way by which the handlers are invoked

Zhu-jiatong, 2023-06-05:
I am working on a project that involves writing to a database file stored on an SD card. A request handler function will write to the database file on once invoked by incoming requests. 
It is suspected that there would be issues, e.g., corruption, when multiple write operations are done simultaneously to the database file. 
So, when there is a large volume of incoming requests in a rather short period of time, how does this library handle them? Does it wait until existing requests are handled, or does it directly interrupt the current ongoing handler function and calls the handler again, or it is some other way?

ToMe25, 2023-06-07:
If by handlers you mean what this library calls "Handlers", I have no clue.
If you, as I assume from context, mean the callback functions for requests, or the response filler callbacks, I might be able to help.

I believe I had an issue with the same method being called twice in parallel on both ESP32 cores, which would mean they can be called in parallel.
This was reasonably possible because I was generating a few mB of output data for a single file download using a callback response.

Even if that was my mistake, and they weren't truly running in parallel, I am pretty certain you can have multiple callbacks for different responses being called alternatingly, to process different requests.

Since the request callbacks are called by a function that gets called by an esp-idf callback, I do not have the know-how to reasonably check whether these are called in parallel or in series, but I believe they can be called in parallel.

Since you also may not delay or yield during such a callback I would, if possible, attempt to completely separate read and write requests, and then have write requests add what should be written to some sort of queue, to be handled by a different Task later.
I am aware this is not always possible, however with the "no delay or yield" function I am not aware of any way to forcefully synchronize request callbacks internally.

Of course one could always return a 500 or 503 response when a new request gets received while another one is currently being handled, but depending on the situation that might not be a valid solution either.

Zhu-jiatong, 2023-06-09:
Thanks. Since it could be in parallel, it would be wise for one to queue the conflicting operations and go through them one by one in the background.
My approach is to dispatch a worker thread for each "shared resource". Each thread will go through a queue of tasks and execute them in order in the background. I made a simple library<https://github.com/Zhu-jiatong/TaskQueue> to abstract this process.
Though it would be great if an official response could be made by the maintainers/authors of this wonderful library to clarify this matter.
[cid:9f48b9df-adde-4dfb-811c-fb75d5917c59]<https://github.com/Zhu-jiatong/TaskQueue>
GitHub - Zhu-jiatong/TaskQueue: A library to serialise tasks from multiple threads<https://github.com/Zhu-jiatong/TaskQueue>
A library to serialise tasks from multiple threads - GitHub - Zhu-jiatong/TaskQueue: A library to serialise tasks from multiple threads
github.com
