me-no-dev/ESPAsyncWebServer #1322: server.on routing

UserNulled, 2023-07-12:
if I make "/system" page and "/system/test" page only "/system" page is working
when using ESP8266Webserver library it was working with no problem.

server->on("/system", HTTP_ANY, [](AsyncWebServerRequest* request) {
  request->send(200, "text/plain", "system");
 });

server->on("/system/test", HTTP_ANY, [this](AsyncWebServerRequest* request) {
   request->send(200, "text/plain", "system test");
 });

is it normal behavior using this library?

Zhu-jiatong, 2023-07-18:
By not working, what do you mean specifically?
1. Web server not responding?
2. The incorrect callback function is invoked?
3. Other behaviour? (please specify)

Zhu-jiatong, 2023-11-10:
I faced the same issue. However, I managed the following workaround:
- Convert all URIs into regex
- Enable compiler flag for regex support

To implement it:
```
// define a function to convert plaintext URIs into regex
String toRegex(const String &uri)
{
	String result = url;
	result.replace("/", "\\/");
	return "^" + result + "(\\?.*)?$";
}

// use the toRegex function when routing
server->on(toRegex("/system").c_str(), HTTP_ANY, [](AsyncWebServerRequest* request) {
	request->send(200, "text/plain", "system");
 });

server->on(toRegex("/system/test").c_str(), HTTP_ANY, [](AsyncWebServerRequest* request) {
	request->send(200, "text/plain", "system test");
 });
```
Remember to add the compiler flag for regex support:
`-DASYNCWEBSERVER_REGEX`

IPdotSetAF, 2024-02-02:
Is there any reason that the `AsyncWebServer` is not using the `Uri` class that comes with `ESP8266WebServer`?
It made URI Definition much easier and cleaner (using UriBraces, UriRegex and such).

I might be able to create a pull request.

closed 2024-02-17 by UserNulled