me-no-dev/ESPAsyncWebServer #1329: How to read POST of multiple select

Mark-81, 2023-07-20:
In my HTML I have a `select` with `multiple` option enabled:

```
<select name="weekday" id="weekday" class="form-select" size="7" multiple aria-label="Weekday">
  <option value="1">Monday</option>
  <option value="2">Tuesday</option>
  <option value="3">Wednesday</option>
  <option value="4">Thursday</option>
  <option value="5">Friday</option>
  <option value="6">Saturday</option>
  <option value="0">Sunday</option>
</select>
```

Here my code to read the POST values:

```
  _server.on("/alarm.html", HTTP_POST, [this](AsyncWebServerRequest *request)
  {
    Serial.println(request->getParam("weekday", true)->value());
    ...
```

It returns only the first item selected. 
How to retrieve the others?

B4stl3r, 2023-07-31:
so you expect multiple values, so an array?
request->getParam()->value() is not an array.. it's a single value ;)
so you have multiple parameters with the same name (as you expect), what did you miss out?

think about using arduinojson and send in your html / js the json object.. works great ;)

Mark-81, 2023-07-31:
> so you have multiple parameters with the same name (as you expect), what did you miss out?

Since we have multiple parameters with the same name, and `getParam()` is not an array, it returns only one of them. I don't understand the right syntax to read them all.

B4stl3r, 2023-07-31:
int paramsNr = request->params();
for (int i = 0; i < paramsNr; i++)

;)

Mark-81, 2023-07-31:
Ok, so my workaround was correct. I was wondering if there is something better than this. Thank!

closed 2023-07-31 by Mark-81