me-no-dev/ESPAsyncWebServer #1330: ESP32-S2 concurrent use of serveStatic from LittleFS and SD card for dataloging not working

wibenjamin, 2023-07-25:
In my project I run the webserver to apply some settings, it is implemented with the serveStatic method:

`server.serveStatic("/", LittleFS, "/www/").setDefaultFile("index.html").setCacheControl("max-age=2592000");`

This works well and stable up until Arduino 2.0.3, all higher Versions seem to break something that the webserver is not responding at all, so i go with 2.0.3 as this Version also got some fixes for SD card.

SD card is working in SPI mode, SD file system is readable, writeable and it is possible to do a firmwareupdate from the card too.
On its own working as expected, perfectly fine.

But bringing those two parts together, after initializing the SD card the webserver is not answering any requests. I really can not make out the reason, even with `-DCORE_DEBUG_LEVEL=5` there are no error messages. The code compiles fine, no errors no warnings.
The LittleFS filesystem definetely is readable from this point, because I load some other data for a different purpose from there.

Did anyone got this LittleFS - SD card constelation working? Is this an issue with this library or do I have to look somewhere else?

Any help or advise is apprechiated.

B4stl3r, 2023-07-31:
Hi, I've similar issues when using LittleFS and SD in the same project.

In addition, LittleFS is (for me) currently extremely slow (SPIFFS is ~5 times faster).
Found also some issues on github stating the same during the last weeks.

I moved back to SPIFFS and everything was working great (also SPIFFS and SD coexistency).

cheers!

wibenjamin, 2023-08-01:
Hi,
thanks for your suggestion, I rolled back to SPIFFS but it is the same problem, as soon both are mounted, no working webserver. Plus if no SD card inserted, serving the web interface from SPIFFS gives time out errors after two or three page reloads without caching. With littleFS this is working flawless and fast for me.

Could you possibly post a code snippet of a known working setup routine for concurrent spiffs and SD card usage?

B4stl3r, 2023-08-01:
hi,
what's your thought? do you want to serve pages from spiffs and sd card? 
from what you wrote, it sounds a little bit strange and i cannot follow you.

i serve pages from spiff and use sd card for other purpose.

wibenjamin, 2023-08-02:
@B4stl3r
With a small html test page I tried something similar like your suggestion and it is working, even served from littleFS and also when the sd card is mounted.

Thank you!

So in general that setup seems to work and it could indeed be an issue somewhere in the serve static function.
I am suspecting there is something going on with the vfs and file paths and ... who knows.
I wont have time to dive deeper here in the next few days and the esp now is dropping the wifi connection ( one step forward, two steps back... ¯\\\_(ツ)\_/¯ ). 

I will give an update if I found a solution or the problem.

wibenjamin, 2023-08-07:
@B4stl3r 
Seemed it was too early, it was only working with a very small text only test page. Anything bigger even its just a few bytes svg, the server stalls, as soon there is the first file request. The rest of the code in the main() loop is still running after that, no reboot, no error messages, its silently dropping the wifi connection (ping not answered from there on) and thats it.

As soon the sd card is not mounted by the system everything working fine again, no issues.

But I feel a little helpless from here on, for me there is no way to debug the code. Its just like digging blindly in the mud, i can not see what is going on here and where the code is falling on its face... :-(

Any ideas?

B4stl3r, 2023-08-07:
hmm.. can be anything... what version of sdk do you use?
did you check your heap?

print e.g. each 5 seconds this:

auto maxHeap = ESP.getMaxAllocHeap();
auto freeHeap = ESP.getFreeHeap();
ESP_LOGD("main", "Free:%i, maxAlloc:%i", freeHeap, maxHeap);

wibenjamin, 2023-08-10:
> hmm.. can be anything... what version of sdk do you use? did you check your heap?
> 
> print e.g. each 5 seconds this:
> 
> auto maxHeap = ESP.getMaxAllocHeap(); auto freeHeap = ESP.getFreeHeap(); ESP_LOGD("main", "Free:%i, maxAlloc:%i", freeHeap, maxHeap);

Thanks, this was a good suggestion!
After mounting the SD-card I dont have enough free allocatable heap memory to feed the web browser requests. Around 30k allocatable heap (without SD-card) seems to be the bare minimum for my setup to work. Mounting the SD-Card with the option of 1 open file only reduces my available memory by 10k. 

Maybe with some optimisations in my code I am able to get it to work. If i can mitigate my heap fragmentation its maybe already enough ( in the moment 50k free heap with 30k allocatable).

B4stl3r, 2023-08-10:
I'm facing similar issues (and i have no debugger to eliminate heap fragmentation or have a more detailed look on it)..
that's why I opened: https://github.com/me-no-dev/ESPAsyncWebServer/issues/1333

I'm serving now gz content, which save's me a little bit, but not much.

wibenjamin, 2023-08-11:
> I'm facing similar issues (and i have no debugger to eliminate heap fragmentation or have a more detailed look on it).. that's why I opened: #1333
> 
> I'm serving now gz content, which save's me a little bit, but not much.

Apperently the IDF offers standalone heap tracing functionality, where you can dump the heap allocation to a log.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#standalone-mode

But to enable this you need to make your Arduino project become an ESP-IDF Project with Arduino as a component. Thats all a little steep for me, seems I got the code itself to compile but I get linker errors.

wibenjamin, 2023-08-11:
To conclude:
It IS possible to use LittleFS and an SD-Card concurrently if you have enough allocatable heap to satisfy the webrequests. The necessary free heap will be depending on your application.

Possible solutions to make it work properly:
- heap optimized code
- ESP Controller with more S-RAM (ESP32 / ESP32-S3) or with PS-RAM ( but that needs some small changes to the library)
- "Heapgently" sending of chunked data to the client ( what ever will be solution to @B4stl3r  issue https://github.com/me-no-dev/ESPAsyncWebServer/issues/1333)

closed 2023-08-11 by wibenjamin

B4stl3r, 2023-08-11:
can you give me more details on:

"But to enable this you need to make your Arduino project become an ESP-IDF Project with Arduino as a component. Thats all a little steep for me, seems I got the code itself to compile but I get linker errors."

? would also like to give it a try..

wibenjamin, 2023-08-11:
> can you give me more details on:
> 
> "But to enable this you need to make your Arduino project become an ESP-IDF Project with Arduino as a component. Thats all a little steep for me, seems I got the code itself to compile but I get linker errors."
> 
> ? would also like to give it a try..

You find some information here:
https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html

And here an IDF example, but it seems that is a little outdated?
https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-blink