me-no-dev/ESPAsyncWebServer #1333: Heap gentle serving of websites

B4stl3r, 2023-08-02:
I've some websites which are served via the Asyncwebserver, including some .css, .js etc..

here you can see the free Heap before calling the website:
[app] main: Free:94388, maxAlloc:69620

during sending of one page (which will also request additionally .css, .js..) via
request->send(SPIFFS, ("/www" + page));
the free (ESP.getFreeHeap()) and maxAlloc Heap (ESP.getMaxAllocHeap()) goes dramatically down:
[app] before sending: Free:89048, maxAlloc:65524
[app] after sending: Free:78320, maxAlloc:51188

[app] before sending: Free:76836, maxAlloc:51188
[app] after sending: Free:65600, maxAlloc:36852

[app] before sending: Free:70636, maxAlloc:42996
[app] after sending: Free:65808, maxAlloc:38900

[app] before sending: Free:65620, maxAlloc:34804
[app] after sending: Free:57096, maxAlloc:27636

so to serve the full site, I require ~ 40kb of heap..
after page **is fully served**, everything goes back to normal:
[app] main: Free:95720, maxAlloc:69620

Already tried **chunked response** (taken from a thread here), but the Heap usage is similar (~2kb less).

      mWeb.on("/test2.html", HTTP_GET, [](AsyncWebServerRequest *request){
      const File SPIFFSfile = SPIFFS.open("/www/setup.html", FILE_READ);
      if (SPIFFSfile){
        Serial.printf("[HTTP]\tSPIFFS File exists [%d]\r\n", SPIFFSfile);
      } else {
        Serial.printf("[HTTP]\tSPIFFS File DOESN'T exists [%d] <<< ERROR !!!\r\n", SPIFFSfile);
      }
      AsyncWebServerResponse *response = request->beginChunkedResponse("text/html", [SPIFFSfile](uint8_t *buffer, size_t maxLen, size_t index) -> size_t {
          File SPIFFSLambdaFile = SPIFFSfile;                                   // Local copy of file pointer
          Serial.printf("[HTTP]\t[%d]\tINDEX [%d]\tBUFFER_MAX_LENGHT [%d]\r\n", index, SPIFFSLambdaFile.size(), maxLen);
          return SPIFFSLambdaFile.read(buffer, maxLen);
        }
      );
      request->send(response);
    }
  );

Any Idea's / Heap gentle code ?

horendus, 2023-08-22:
Use 7-zip to compress your web files to gzip format and serve those instead for a great reduction in transfer size and ram usage.

If having issues on iPhone rename the files from JavaScript.jz.gz to JavaScript.jsgz. Remove the extra .

B4stl3r, 2023-08-22:
that's what I already have in place ;)

Any other hints?