me-no-dev/ESPAsyncWebServer #1341: [REQUEST] Serve files from static PROGMEM declarations

bmedici, 2023-08-18:
The most used case is of course, to serve static files from a SPIFFS filesystem.
This is great and works fine, for example with
```
server.serveStatic("/", SPIFFS, "/www/");
```

In my case, I need to embed all the served files into the firmware. Many reasons for these reasons
- i can't really build a separate binary filesystem image and ship it separately
- having the files along the source code ensures that web pages/assets match the exact version of code running (always synchronised)
- storing web content within the code allows me to save space in flash by merging code and data storage (a few kB that make the difference)
- I already have a complex build script step, building html pages from templates, gzipping assets to minimise space

This build phase already builds a .h file containing PROGMEM declarations for theses files :
```
static const char file1[] PROGMEM = {
0x3c, 0x68, 0x74, 0x6d, 0x6c, 0x20, 0x6c, 0x61, 0x6e, 0x67, 0x3d, 0x22, 0x65, 0x6e, 0x22, 0x3e, 0xa, 0x20, 0x20, 0x3c, 
0x68, 0x65, 0x61, 0x64, 0x3e, 0xa, 0x20, 0x20, 0x20, 0x20, 0x3c, 0x6d, 0x65, 0x74, 0x61, 0x20, 0x63, 0x68, 0x61, 0x72, 
```

These are compiled along with the code and can be accessed by the webserver as-is.

I wish we could have something like SPIFFS interface to serve these assets from an index, for example a std::unordered_map with the file path as the key, and file contents (or pointer to data array) as the value.
Having the webserver handle the "not found" situation. Plus, I have use a regex path to get the path and serve the content, which is less than optimal.

Something like 
```
server.serveStatic("/", MyStaticHandler, "/www/");
```
could be great, and we could implement simple functions for a lookup in the map, to provide the right content-type and of course, the content pointer to return.

The implementation could automatically handle the file compression (or de-compression) as it works today for SPIFFS files. As long as we provide gzipped-bytes-array as the data and tell the webserver the content is gzipped.

Any thoughts about this ?

Thanks for this great projet, and for your suggestions.

hoeken, 2023-10-25:
Something like this would be great.  I went down the SPIFFS route as well at first - nice and easy for development, but doesn't work well for doing OTA updates.  Having it all in one single binary is definitely the way to go.

bmedici, 2023-10-25:
Any plan to implement this within the webserver maybe ? @me-no-dev