me-no-dev/ESPAsyncWebServer #1346: EspAsyncWebServer creates incorrect size header for chunks return. AioHttp catches this error

ghisbo, 2023-10-02:
When using aioHttp in Python to read the content of a ESP8266 webserver, the program stops with an error " incorrect data in chunk size. In fact there is a \r. 
Http.client and Postman are more permissive and allow for this error. The problem with Http.client is that after a bigger number of request, the program hangs.
So aioHttp is the way to go but this discovers and reports the error.`
I insert the code for the webserver, this comes from the example 
( see the /getVector )`for the code that provokes the error

webserver code:
`
//
// A simple server implementation showing how to:
//  * serve static messages
//  * read GET and POST parameters
//  * handle missing pages / 404s
//

#include <Arduino.h>
#ifdef ESP32
#include <WiFi.h>
#include <AsyncTCP.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#endif
#include <ESPAsyncWebServer.h>
#include <vector>
#include <stdio.h>

AsyncWebServer server(80);

const char *ssid = "WiFi-2.4-F808";
const char *password = "wn52mbs2h3pkf";

const char *PARAM_MESSAGE = "message";

// keep track of chunks ( instead of index in a file or string)
static int chunkCounter = 0;
int vectorSize = 2;

void notFound(AsyncWebServerRequest *request)
{
    request->send(404, "text/plain", "Not found");
}
std::string a = "0123456789012345678901234567890123456789";
std::string b = "abcdefghijabcdefghijabcdefghijabcdefghij";

std::string myArray[] = {a, b};

size_t awsResponseFiller(uint8_t *buffer, size_t maxLen, size_t index)
{
    // index is the returned position in a long string
    // copy parts from that string in the buffer

    

    if (chunkCounter < vectorSize)
    {
        int bufLen=myArray[chunkCounter].size();
        std::memcpy(buffer, (myArray[chunkCounter]).c_str(), bufLen); // copy full string length
        chunkCounter++;
        return bufLen;
    }

    buffer[0] = 0;
    // Serial.println("buffer 0");
    return 0;
}

void setup()
{

    Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    if (WiFi.waitForConnectResult() != WL_CONNECTED)
    {
        Serial.printf("WiFi Failed!\n");
        return;
    }

    Serial.print("IP Address: ");
    Serial.println(WiFi.localIP());

    server.on("/", HTTP_GET, [](AsyncWebServerRequest *request)
              { request->send(200, "text/plain", "Hello, world"); });

    server.on("/getVector", HTTP_GET, [](AsyncWebServerRequest *request)
              {
                  chunkCounter = 0;

                  AsyncWebServerResponse *response = request->beginChunkedResponse(
                      "application/json", awsResponseFiller);

                  response->addHeader("Server", "ESP Async Web Server");

                  request->sendChunked("application/json", awsResponseFiller);
              });

    // Send a GET request to <IP>/get?message=<message>
    server.on("/get", HTTP_GET, [](AsyncWebServerRequest *request)
              {
        String message;
        if (request->hasParam(PARAM_MESSAGE))
        {
            message = request->getParam(PARAM_MESSAGE)->value();
        }
        else
        {
            message = "No message sent";
        }
        request->send(200, "text/plain", "Hello, GET: " + message); });

    // Send a POST request to <IP>/post with a form field message set to <message>
    server.on("/post", HTTP_POST, [](AsyncWebServerRequest *request)
              {
        String message;
        if (request->hasParam(PARAM_MESSAGE, true))
        {
            message = request->getParam(PARAM_MESSAGE, true)->value();
        }
        else
        {
            message = "No message sent";
        }
        request->send(200, "text/plain", "Hello, POST: " + message); });

    server.onNotFound(notFound);

    server.begin();
}

void loop()
{
}
`
Python aioHttp

`
```
# basic example for chunks with aiohttp
# this generates an error
# Exception has occurred: ClientResponseError
# 400, message="Invalid character in chunk size:\n\n  b'28  \\r'\n      ^", 
# url=URL('http://192.168.1.27:80/getVector')
# the construction of the chunk is not correct ( chunk size contains \r)
# 
```
import aiohttp
import asyncio

async def main():

    async with aiohttp.ClientSession("http://192.168.1.27:80") as session:

        for i in range(10):
            async with session.get("/getVector") as resp:
                print(resp.headers)
                print(resp.status)
                print(resp.url)
                # print(await resp.text())
                async for chunk in resp.content.iter_chunks():
                    print(chunk)
  

if __name__ == "__main__":
    asyncio.run(main())

`

HttpClient version

`
```
# basic example for chunks with http.client
# this works OK for a number of vectors but after a while 
# the program hangs
# output
# Chunk Received: 0123456789012345678901234567890123456789
# Chunk Received: abcdefghijabcdefghijabcdefghijabcdefghij
# 
```
import http.client

def main():

    host="192.168.1.27:80"
    port=80
    conn= http.client.HTTPConnection(host)

    conn.request("GET","/getVector", headers={"Host": host})

    resp = conn.getresponse()
    resp.chunked = False

    def get_chunk_size():
        size_str = resp.read(2)
        while size_str[-2:] != b"\r\n":
            size_str += resp.read(1)
        return int(size_str[:-2], 16)

    def get_chunk_data(chunk_size):
        data = resp.read(chunk_size)
        resp.read(2)
        return data

    respbody = ""
    while True:
        chunk_size = get_chunk_size()
        if (chunk_size == 0):
            break
        else:
            chunk_data = get_chunk_data(chunk_size)
            print("Chunk Received: " + chunk_data.decode())
            respbody += chunk_data.decode()

    conn.close()
    print(respbody)

if __name__ == "__main__":
    main()
    

`

ghisbo, 2023-10-02:
I found the error in the Webresponse.cpp: in "size_t AsyncAbstractResponse::_ack(AsyncWebServerRequest *request, size_t len, uint32_t time)", line 320 should be "%04x" instead of "%x"
`
outLen = sprintf((char*)buf+headLen, "%04x", readLen) + headLen;
`
I do not know how to adapt the library here.

jetcat-bob, 2023-10-04:
> I found the error in the Webresponse.cpp: in "size_t AsyncAbstractResponse::_ack(AsyncWebServerRequest *request, size_t len, uint32_t time)", line 320 should be "%04x" instead of "%x" `outLen = sprintf((char*)buf+headLen, "%04x", readLen) + headLen;` I do not know how to adapt the library here.

That worked for me. Good find!!!! So, I guess you must pad outLen string with zeros instead of spaces???

ghisbo, 2023-10-04:
Yes, correct.
However, this solves part of my problem. When i run the the loop in python, after +- 120 requests, there is an error: 

> Count=  119  
 Error Message:  400, message="Expected HTTP/:\n\n  bytearray(b'0028\\r')\n               ^", url=URL('http://192.168.1.16:80/getVector')

this looks as if the chunk length is causing an error ( the \r is probably not wanted) but i cannot find where it comes from.
Maybe you can run the same test and see if the program crashes

jetcat-bob, 2023-10-04:
I just did a loop of 1252 times transferring around 1400 bytes per chunk
and it worked fine. I'm using c++ with the Arduino framework.

ghisbo, 2023-10-04:
Ok that looks good!. I will have to check my setup all over? 
thanks for testing and for your feedback

ghisbo, 2023-10-04:
What esp8266 are you using. I use a esp8266 d1 mini and it keeps producing erros ( to test i switched from vscode to arduino ide
and i used also a new esp8266.

jetcat-bob, 2023-10-04:
Using a ESP32-S3 on unexpected makers board.
I'm using the Aysnc library for uploading & downloading files in my data
logger program that can produce large files. If I uploaded a file > 5.6k,
it would stop the call backs to my handler. When I made the formatting
change to the outLen string, it works perfectly fine. My program still will
not download more than 5.6k but I think that's because my download handler
is not setup correctly for sending the file in chunks. I'm in the process
of trying to figure it out.
I'm just beginning to get a handle on this library as it is so I'm probably
not much help. I'm happy to test things on my end for you though.

Bob Wilcox

ghisbo, 2023-10-05:
I have gone back to the Esp8266Webserver ( not async ) and that is stable. The EspAsyncWebServer is not stable enough. I need a stable server, running for long times (IOT). 
With the regular web server i have a slower but stable solution.

jetcat-bob, 2023-10-05:
I'm going to give it one more try today and if not successfully, I'll go
back to the non async version myself. I have that one working perfectly.
I think the authors of this library are perusing other things by the date
of the last updates.
I was impressed on the bug you did find!

Bob Wilcox

ghisbo, 2023-10-06:
Thx for your help. I close it by now

closed 2023-10-06 by ghisbo