me-no-dev/ESPAsyncWebServer #1352: error: 'mbedtls_md5_starts_ret' was not declared in this scope; did you mean 'mbedtls_md5_starts'?

l-pastor, 2023-10-17:
Hi,

I use lastest library version on Arduino IDE 2.2.1, with ESP32 wroom32, when I compile I get many errors.

First lines starting with : 

d:\Mega\� Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp: In function 'bool getMD5(uint8_t*, uint16_t, char*)':
d:\Mega\� Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp:74:3: error: 'mbedtls_md5_starts_ret' was not declared in this scope; did you mean 'mbedtls_md5_starts'?
   74 |   mbedtls_md5_starts_ret(&_ctx);
.....

Best regards,
Laurent

blackhack, 2023-10-19:
Apply this pull request to the lib: https://github.com/me-no-dev/ESPAsyncWebServer/pull/1351

Or use this fork for now while @me-no-dev has the opportunity to review this repo: https://github.com/blackhack/ESPAsyncWebServer.git 
Besides from the build errors, I fixed a crash that most likely will affect you.

l-pastor, 2023-10-21:
Thank you very much blackhack, you did a good job.

aly-fly, 2024-02-10:
Using recent PlatformIO with ESP32 core v2. 
Got an error in WebAuthentication.cpp:
`mbedtls_md5_starts(&_ctx);` does not exist.
Replaced with `mbedtls_md5_starts_ret(&_ctx);` and it compiles fine.

mali119, 2024-10-11:
I got this problem when building WebSocket server on ESP32. It turned out I installed the wrong library.  So just change the library, problem solved