me-no-dev/ESPAsyncWebServer #1360: Difficulty Implementing SSL with ESP32 - Seeking Assistance

maxklt, 2023-11-14:
Hello @me-no-dev 👋

I hope this message finds you well. I am currently working on a project that involves using the ESPAsyncWebServer library with SSL for secure communication. However, I have been encountering significant challenges in getting it to work with HTTPS.

I have referred to the documentation and various online resources, but the process seems to be more complex and less straightforward than anticipated. I was wondering if you could provide some guidance or share any insights on how to effectively implement SSL with the ESPAsyncWebServer library.

Additionally, it would be greatly appreciated if you could highlight any specific considerations, steps, or configurations that might streamline the process of setting up HTTPS. It seems to be more challenging than expected, and any assistance you could offer would be invaluable.

Thank you for your time and consideration. I look forward to your guidance on this matter.

Best regards, Mex

maxklt, 2023-11-21:
@me-no-dev Would be greate to get any help regarding this topic, this problem keeps cropping up in the community

LukasHNZ, 2023-11-24:
> @me-no-dev Would be greate to get any help regarding this topic, this problem keeps cropping up in the community

If a solution pops up I would be very interested aswell!