me-no-dev/ESPAsyncWebServer #1366: esp32 reboot when i use a "for" function with hasParam and getParam inside then.

adolfomcs, 2023-12-14:
Hi guys, anyone can help me with my code?
In my program, i wrote a routine inside a route block, that use "for function" to automate the coding process and get several values from web client. But i have a reboot problem when the "for function" reaches the ninth iteration.

my routine that reboot the esp32:
``` cpp
      // novo cronograma
      server.on("/new_crono_config", HTTP_GET, [] (AsyncWebServerRequest *request) {

        uint8_t index_cron = 255;

        // número de indexador
        if (request->hasParam("matriz_index")) index_cron = request->getParam("matriz_index")->value().toInt();

        if (index_cron != 255) {

          // horário de início
          if (request->hasParam("prg_hr_i")) HorarioIni[index_cron][0] = request->getParam("prg_hr_i")->value().toInt();
          if (request->hasParam("prg_mn_i")) HorarioIni[index_cron][1] = request->getParam("prg_mn_i")->value().toInt();

          // partidas
          for (uint8_t count = 1; count <= 16; count++) {

            char param_num[2] = "\0";
            sprintf(param_num, "%d", count);

            // setor
            char param[10] = "\0";
            strcat(param, "p");
            strcat(param, param_num);
            strcat(param, "_str");
            if (request->hasParam(param)) Setor[index_cron][count] = request->getParam(param)->value().toInt();

            // tempo programa
            memset(param, '\0', sizeof(param));
            strcat(param, "p");
            strcat(param, param_num);
            strcat(param, "_tmp");
            if (request->hasParam(param)) ProgTempo[index_cron][count] = request->getParam(param)->value().toInt();

            // tempo químico 1
            memset(param, '\0', sizeof(param));
            strcat(param, "p");
            strcat(param, param_num);
            strcat(param, "_qm1");
            if (request->hasParam(param)) Qm1Tempo[index_cron][count] = request->getParam(param)->value().toInt();

            // tempo químico 2
            memset(param, '\0', sizeof(param));
            strcat(param, "p");
            strcat(param, param_num);
            strcat(param, "_qm2");
            if (request->hasParam(param)) Qm2Tempo[index_cron][count] = request->getParam(param)->value().toInt();
          }

          // horário de término
          if (request->hasParam("prg_hr_f")) HorarioFim[index_cron][0] = request->getParam("prg_hr_f")->value().toInt();
          if (request->hasParam("prg_mn_f")) HorarioFim[index_cron][1] = request->getParam("prg_mn_f")->value().toInt();
        }

        request->send(200, "text/html", "<body onload=\"rtrn()\"> <script> function rtrn() { location.assign(\"/cnf_cron7SAB2\"); } </script> </body>");
      });
```

as i mentioned before, when the for function reaches the ninth iteration, the esp32 reboot and give me that in the serial:

ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4

no backtrace, just reboot.

Anyone can explain me why this happen?
Thanks!

adolfomcs, 2023-12-14:
if i comment the lines that have the methods hasParam and getParam, the for function iterations happen normaly.

MichaelDvP, 2024-01-10:
`char param_num[2] = "\0";` is to short to hold numbers with 2 digits and terminator. Just make it [3].