me-no-dev/ESPAsyncWebServer #1379: support of SdFat library

JoergTiedemann, 2024-02-19:
Is there any solution to support hosting files from an sd card using sd_fat library ?
in request->send first parameter is fs::FS or File but not FsFile which is used by SdFat Library

Red-Owl, 2024-06-06:
#include "SD.h"
#include "FS.h"

server.on("/favicon.ico", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SD, "/favicon.ico", "image/x-icon");
});

VeloSteve, 2024-09-12:
> #include "SD.h" #include "FS.h"

A solution using SD.h does not help.  I get much higher error rates using SD.h.

My current workaround is

1. For static files I put them in the SPIFFS filesystem before installing the main program and use something like `server.serveStatic("/", SPIFFS, "/htdocs/").setCacheControl("max-age=86400");`
2. For files on the SD card, I use `request->sendChunked()` and call my own function which reads up to 4096 bytes at a time using SdFat.

Of course either half of the workaround could work for all files, but this work for me.  It is partly because my static files are mostly small and the only file I normally send from the SD card can be very large.