me-no-dev/ESPAsyncWebServer #1382: POST not accepting content type: application/x-www-form-urlencoded

IPdotSetAF, 2024-02-22:
POST endpoint does not accept content type: `application/x-www-form-urlencoded`

this content type is used for form data submission in html.

because we are using POST request the form data will not be provided in the query params and will be in the request body, but even the body handler does not execute with this content type.

IPdotSetAF, 2024-02-22:
looks like it only accepts `application/json` as `content type`!!!!

danieljharris, 2024-04-04:
For whatever reason `application/x-www-form-urlencoded` is treated differently to all other content types: https://github.com/me-no-dev/ESPAsyncWebServer/blob/f71e3d427b5be9791a8a2c93cf8079792c3a9a26/src/WebRequest.cpp#L146