me-no-dev/ESPAsyncWebServer #1399: where i can get request if i mast send JSON to all connected clients by timer ?! no good exapmles no information!

sova32, 2024-04-23:
#include "AsyncJson.h"
#include "ArduinoJson.h"

AsyncJsonResponse * response = new AsyncJsonResponse();
response->addHeader("Server","ESP Async Web Server");
JsonObject& root = response->getRoot();
root["heap"] = ESP.getFreeHeap();
root["ssid"] = WiFi.SSID();
response->setLength();
request->send(response);

where i can get request if i mast send JSON to all connected clients by timer ?! no good exapmles no information!
where i must put this code? in asshole?

Pablo2048, 2024-04-23:
Don't be rude n00b - you did nothing for the community as the author of this library did. To answer your question:
There is nothing like 'connected client' in pure http context at least if you are not talking about websockets. If you are talking about websockets, here is something to read: https://github.com/me-no-dev/ESPAsyncWebServer?tab=readme-ov-file#methods-for-sending-data-to-a-socket-client you can also STFW ( https://m1cr0lab-esp32.github.io/remote-control-with-websocket/websocket-and-json/ ).

sova32, 2024-04-27:
> Don't be rude n00b - you did nothing for the community as the author of this library did. To answer your question: There is nothing like 'connected client' in pure http context at least if you are [...]

sorry but its dont help me. i make websocket server to send data from esp to all clients and AsyncCallbackJsonWebHandler for get json. why web socket server reboot esp after get json and send response i dont know.
but documentation is bad. i lost 2 weeks.

closed 2024-04-27 by sova32