me-no-dev/ESPAsyncWebServer #1404: AsyncWebServerRequest Questions

acztassi, 2024-04-29:
hello.

I'm building a little framework for manage a complex automation, then the URIs are not gonna be the same every time..
then I was intending to use AsyncWebServerRequest, but I really didn't understand if is it possible .. let me show the idea

class NewHandler: public AsyncWebServerRequest
{
public: 
       bool canHandle(AsyncWebServerRequest *request) override {
             return (request->url() == "/endpoint1" || request->url() == "/endpoint2");
       }  
        virtual esp_err_t handleRequest(AsyncWebServerRequest *request) override{
               if (request->url() == "/endpoint1")
                   { 
                           return request->reply(200, "text/html", "called endpoint 1");
                    }
               else
               if (request->url() == "/endpoint2")
                   { 
                           return request->reply(200, "text/html", "called endpoint 2");
                    }
              else
                    {
                            return request->reply(404, "text/html", "endpoint not managed");
                     }
        } 

}

only add this class and do not use "server.on(...." to set endpoints
is it possible? Is there any issue?

CelliesProjects, 2024-05-07:
You want to update the url handlers is my guess.
Maybe something like this?

```c++
void updateFileHandlers(const tm &now)
{
    static char path[100];
    snprintf(path, sizeof(path), "/%i/%i/%i.log", now.tm_year + 1900, now.tm_mon + 1, now.tm_mday);

    log_d("Current logfile: %s", path);

    static AsyncCallbackWebHandler *currentLogFileHandler;
    http_server.removeHandler(currentLogFileHandler);
    currentLogFileHandler = &http_server.on(path, HTTP_GET, [](AsyncWebServerRequest *const request)
                                            {
        if (!SD.exists(path)) return request->send(404);
        AsyncWebServerResponse* const response = request->beginResponse(SD, path);
        response->addHeader(CACHE_CONTROL_HEADER, CACHE_CONTROL_NOCACHE);
        request->send(response);
        log_d("Request for current logfile"); });

    static AsyncStaticWebHandler *staticFilesHandler;
    http_server.removeHandler(staticFilesHandler);
    staticFilesHandler = &http_server.serveStatic("/", SD, "/").setCacheControl("public, max-age=604800, immutable");
}
```

Called like this:
```c++
    static tm now;
    getLocalTime(&now);
    if ((59 == now.tm_sec) && !(now.tm_min % SAVE_TIME_MIN) && (numberOfSamples > 2))
        saveAverage(now);

    static uint8_t currentMonthDay = now.tm_mday;
    if (currentMonthDay != now.tm_mday)
    {
        updateFileHandlers(now);
        currentMonthDay = now.tm_mday;
    }
```

This code updates the log file handler every time a new day starts.

[Some context ](https://github.com/CelliesProjects/smartMeterLogger-pio/blob/main/src/smartMeterLogger-esp32.ino#L56C1-L76C2)

acztassi, 2024-05-07:
the problem i that I can't register handlers trough "server.on( ...", because the uris won't be the same every time .. for a better comprehension, there will be one main uri, like "/apiconfig" where i will register or remove addresses from the webserver. Then I was intending to do not register handlers, but use only one to manage all urls