me-no-dev/ESPAsyncWebServer #1407: Question: is it possible to replicate AsyncWebSocket with WebServer in arduino-esp32 v3.0.0 ?

PaulZC, 2024-05-22:
Hi All,

We have been using this fantastic library very successfully for a couple of years, with arduino-esp32 <= v2.0.11.

I am investigating whether it is now possible to replace this library with [WebServer](https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer/examples) on arduino-esp32 v3.0.0 (RC3). I have the WebServer running, but I can not find an equivalent for AsyncWebSocket and/or a handler for "/ws"...

Is it possible to replicate AsyncWebSocket with WebServer in arduino-esp32 v3.0.0 ?

I'd be really grateful if someone could point me at some code....

Sincere thanks in advance,
Paul

noweare1, 2024-05-23:
WebServer does not support webSockets. There is a websocket library I used in one of my projects located [here](https://github.com/Links2004/arduinoWebSockets) It worked well. 
Why don't you want to use espAsynchWebserver library for websockets ?

PaulZC, 2024-05-24:
Hi Joseph (@noweare1 ),

This is a great library, it works really well. With arduino-esp32 v3.0.0, there is issue #1391 - but that can be solved. I'm investigating if it is now possible to achieve the same functionality using only the latest espressif core - without needing additional libraries (this and AsyncTCP included). My solution is to use WebServer and esp_http_server simultaneously. esp_http_server is only serving the /ws websocket - on port 81. WebServer is doing everything else - on port 80.

https://www.esp32.com/viewtopic.php?f=19&t=39987

Best wishes,
Paul

closed 2024-05-24 by PaulZC

noweare1, 2024-05-24:
I didn't know esp_http_server supported websockets. Like you, I try to keep the external libraries to a minimum and try to write my own code if I am up to it. Webservers/Websockets is way above my pay grade though : ).  will have to take a look at that library. Since it is part of the IDF it is much better documented with lots of examples. Do you use esprressif-arduino or IDF framework ?

PaulZC, 2024-05-24:
"Do you use esprressif-arduino or IDF framework ?" arduino-esp32

v3.0.0 includes full support for the Wiznet W5500 Ethernet PHY under Arduino. It works very well with other SPI devices - microSD included.