me-no-dev/ESPAsyncWebServer #1408: AsyncWebServer Questions

kRYstall9, 2024-05-22:
Hi,
I'm trying to build a project that uses the esp 32 wrover kit (with camera module) to take jpeg frames and send it to a flask server that does all the detection stuff. Everything seems promising but, whenever I start the loop in order to send the frames, after a while the esp 32 cam crashes and logs this error:

E (700986) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (700986) task_wdt:  - async_tcp (CPU 0/1)
E (700986) task_wdt: Tasks currently running:
E (700986) task_wdt: CPU 0: IDLE0
E (700986) task_wdt: CPU 1: IDLE1
E (700986) task_wdt: Aborting.

abort() was called at PC 0x400ed069 on core 0

Backtrace: 0x400839a9:0x3ffbecec |<-CORRUPTED

ELF file SHA256: 9fdf92501b706253

Rebooting...

I'll leave my code below:

```

void onFrameRequest(AsyncWebServerRequest * request);
void getFrame(AsyncClient* client);

String ssid;
String password;
AsyncWebServer asyncServer(8000);
size_t _jpg_buf_len = 0;
uint8_t *_jpg_buf = NULL;
uint32_t curFrameTime = 0;
uint32_t prevFrameTime = 0;

void setup() {
  Serial.begin(460800);
  
  if(!SPIFFS.begin(true)){
    Serial.println("Failed to initialize SPIFFS.");
    return;
  }
  while(WiFi.begin(ssid, password) != WL_CONNECTED){ Serial.print(".")}
  if(camera_init() != ESP_OK)
    return;
  
  
  
  Serial.print("Camera Ready! Use 'http://");
  Serial.print(WiFi.localIP());
  Serial.println("' to connect");

  asyncServer.on("/stream", onFrameRequest);
  asyncServer.begin();
}

void loop() {
  delay(1);
}

void getFrame(AsyncClient* client){
  camera_fb_t* fb = NULL;
  esp_err_t err = ESP_OK;
  uint32_t curFrameTime =0;
  
  while(true){
    err = ESP_OK;
    fb = esp_camera_fb_get();
    if(fb){
      //Serial.println("Took the frame");
      if(fb->format != PIXFORMAT_JPEG){
        //Serial.println("Frame not in jpeg");
        bool converted = frame2jpg(fb, 80, &_jpg_buf, &_jpg_buf_len);
        esp_camera_fb_return(fb);
        fb = NULL;
        if(!converted){
          err = ESP_FAIL;
        }
      }
      else{
        //Serial.println("Frame already in jpeg");
        _jpg_buf = fb->buf;
        _jpg_buf_len = fb->len;
      }
    }
    else{
      err = ESP_FAIL;
      //Serial.println("Failed buffer");
    }
    if(err == ESP_OK){
      int foundffd9 = 0;
      for (int j=1; j<= 1025; j++){
        if(_jpg_buf[_jpg_buf_len - j] == 0xD9){
          if(_jpg_buf[_jpg_buf_len - j - 1] == 0xFF){
            if(j==1){
              Serial.println("Normal jpeg");
            }
            else{
              Serial.println("Extended jpeg");
            }
            foundffd9 = 1;
            break;
          }
        }
      }
      if(!foundffd9){
        Serial.println("bad jpeg");
        esp_camera_fb_return(fb);
      }
      else{
        //Serial.println("Sending frame to the client");
        if(client->connected())
          client->write((char*)_jpg_buf, _jpg_buf_len);
      }   
    }
    curFrameTime = millis();
    prevFrameTime = (curFrameTime-prevFrameTime);
    double fps = 1/prevFrameTime;
    Serial.printf("%f fps\n", fps);
    prevFrameTime = curFrameTime;
    if(fb){
      esp_camera_fb_return(fb);
      _jpg_buf = NULL;
    }
    else{
      //Serial.println("Frame already null");
      free(_jpg_buf);
    }
    if(client->disconnected()){
      break;
    }
    delay(1);
  }
}

void onFrameRequest(AsyncWebServerRequest * request){
  getFrame(request->client());
}

```

noweare1, 2024-05-23:
This happens when a task is taking up too many cpu cycles. The doc that describes the operation of the esp32 watch dog timer is [here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/wdts.html)

The solution probably will be using "esp_task_wdt_reset()" in the task you assigned to prevent the TWDT from elapsing its timeout period.

kRYstall9, 2024-05-24:
Hi @noweare1, thanks for answering. I've already tried adding the esp_task_wdt_reset() inside the callback, but it didn't help. Since the connection will be closed on request->send(), right now I have no idea on how to implement something that can help me achieving the live streaming. With the WebServer.h library(The one not async) I've been able to achieve that, but it takes too much time to send frames through server.send() or server.sendContent()

noweare1, 2024-05-24:
I am just learning to use the espAsynchWebServer library. I am also trying to stream frames using this library. It took me a while to figure out how streaming works with the espressif Camera example. 

Some of the magic is actually in the header files. The header  "multipart/x-mixed-replace;boundary=" PART_BOUNDARY" is part of the response to  the browser. That header tells the browser that it will pushing images to the browser without having the browser ask for the next image (which saves time). So you might want to take a look at that.

kRYstall9, 2024-05-24:
Hi @noweare1 , thanks again for answering. I took a look at some examples I found and came up to this.

```
#define PART_BOUNDARY "123456789000000000000987654321"
static const char* STREAM_CONTENT_TYPE = "multipart/x-mixed-replace;boundary=" PART_BOUNDARY;
static const char* STREAM_BOUNDARY = "\r\n--" PART_BOUNDARY "\r\n";
static const char* STREAM_PART = "Content-Type: %s\r\nContent-Length: %u\r\n\r\n";

static const char * JPG_CONTENT_TYPE = "image/jpeg";

void handleStreaming(AsyncWebServerRequest *request){
    AsyncWebServerResponse* response = request->beginChunkedResponse(JPG_CONTENT_TYPE, _sendContent);
    response->addHeader("Access-Control-Allow-Origin", "*");
    request->send(response);
}

size_t _sendContent(uint8_t* buffer, size_t maxLen, size_t index){
    Serial.println("MAX LEN: " + String(maxLen));
    Serial.println("INDEX: " + String(index));
    Serial.println("STRLEN BOUNDARY: " + String(strlen(STREAM_BOUNDARY)));
    Serial.println("STRLEN PART: " + String(strlen(STREAM_PART)));
    Serial.println("STRLEN CONTENT TYPE: " + String(strlen(JPG_CONTENT_TYPE)));

    esp_err_t err = ESP_OK;
    vTaskDelay(10);
    if(!_frame.fb || _frame.index == _jpg_buf_len){
        Serial.println("Need a new frame");
        //Ho bisogno di un nuovo frame
        if(index && _frame.fb){
            //Ho gia' inizializzato il frame almeno una volta
            if(_frame.fb->format != PIXFORMAT_JPEG){
                Serial.println("Previous frame was not in JPEG format");
                free(_jpg_buf);
            }
            esp_camera_fb_return(_frame.fb);
            _frame.fb = NULL;
            _jpg_buf_len = 0;
            _jpg_buf = NULL;
        }
        if(maxLen < (strlen(STREAM_BOUNDARY) + strlen(STREAM_PART) + strlen("image/jpeg") + 8)){
            Serial.println("Cannot send the frame. Retry!");
            return RESPONSE_TRY_AGAIN;
        }
        //Getting new frame
        _frame.index = 0;
        Serial.println("Getting a new frame");
        _frame.fb = esp_camera_fb_get();
        if(_frame.fb == NULL){
            Serial.println("Camera failed");
            if(esp_camera_deinit() == ESP_OK){
                camera_init();
            };
            return 0;
        }

        if(_frame.fb->format != PIXFORMAT_JPEG){
            Serial.println("new frame not in JPEG format");
            bool jpeg_converted = frame2jpg(_frame.fb, 80, &_jpg_buf, &_jpg_buf_len);
            if(!jpeg_converted){
                Serial.println("Conversion failed");
                _frame.fb = NULL;
                _jpg_buf_len = 0;
                _jpg_buf = NULL;
                return  0;
            }
            esp_task_wdt_reset();
        }
        Serial.println("New frame's already in JPEG format");
        Serial.println("Sending boundary");
        size_t boundaryLen = 0;
        if(index){
            boundaryLen = strlen(STREAM_BOUNDARY);
            memcpy(buffer, STREAM_BOUNDARY, boundaryLen);
            buffer += boundaryLen;
        }
        //send header
        Serial.println("Sending header");
        size_t headerLen = sprintf((char*)buffer, STREAM_PART, "image/jpeg", _jpg_buf_len);
        buffer += headerLen;
        //send frame
        headerLen = maxLen - headerLen - boundaryLen;
        Serial.println("Sending frame");
        if(headerLen > _jpg_buf_len){
            maxLen -= headerLen - _jpg_buf_len;
            headerLen = _jpg_buf_len;
        }
        memcpy(buffer, _jpg_buf, headerLen);
        _frame.index += headerLen;
        return maxLen;
    }
    Serial.println("Sending frame 2");
    size_t available = _jpg_buf_len - _frame.index;
    if(maxLen > available){
        maxLen = available;
    }
    memcpy(buffer, _jpg_buf+ _frame.index , maxLen);
    _frame.index += maxLen;
    esp_task_wdt_reset();
    return maxLen;
}
```

I can see in the console that it is indeed working, but:

- I'm not able to see the image on the browser page
- Still getting Task watchdog got triggered after a while

EDIT:
I forgot to assign _frame.fb->buf to _jpg_buf and _frame.fb->len to _jpg_buf_len. But still, I'm not able to see the images on a web page

kRYstall9, 2024-05-27:
@noweare1 I managed to solve the problem. I had to create a class that extends the AsyncAbstractResponse class. Then I had to override the _fillBuffer method and it happened to be working.
```
#include "AsyncJpegStreamResponse.h"
#include <cstring>

AsyncJpegStreamResponse::AsyncJpegStreamResponse() {
    _callback = nullptr;
    _code = 200;
    _contentLength = 0;
    _contentType = STREAM_CONTENT_TYPE;
    _sendContentLength = false;
    _chunked = true;
    _index = 0;
    _jpg_buf_len = 0;
    _jpg_buf = NULL;
    memset(&_frame, 0, sizeof(camera_frame_t));
}

AsyncJpegStreamResponse::~AsyncJpegStreamResponse() {
    Serial.println("In the constructor");
    if (_frame.fb) {
        Serial.println("Frame fb not null");
        if (_frame.fb->format != PIXFORMAT_JPEG) {
            free(_jpg_buf);
        }
        esp_camera_fb_return(_frame.fb);
    }
}

bool AsyncJpegStreamResponse::_sourceValid() const {
    return true;
}

size_t AsyncJpegStreamResponse::_fillBuffer(uint8_t* buf, size_t maxLen) {
    Serial.println("In the _fillBuffer function");
    size_t ret = _sendContent(buf, maxLen, _index);
    if (ret != RESPONSE_TRY_AGAIN) {
        _index += ret;
    }
    return ret;
}

size_t AsyncJpegStreamResponse::_sendContent(uint8_t* buffer, size_t maxLen, size_t index) {
    Serial.println("In the _sendContent function");
    if (!_frame.fb || _frame.index == _jpg_buf_len) {
        if (index && _frame.fb) {
            if (_frame.fb->format != PIXFORMAT_JPEG) {
                free(_jpg_buf);
            }
            esp_camera_fb_return(_frame.fb);
            _jpg_buf = NULL;
            _jpg_buf_len = 0;
            _frame.fb = NULL;
        }

        if (maxLen < (strlen(STREAM_BOUNDARY) + strlen(STREAM_PART) + strlen(JPG_CONTENT_TYPE) + 8)) {
            return RESPONSE_TRY_AGAIN;
        }

        _frame.index = 0;
        _frame.fb = esp_camera_fb_get();
        if (_frame.fb == NULL) {
            log_e("Camera frame failed");
            return 0;
        }

        if (_frame.fb->format != PIXFORMAT_JPEG) {
            bool _jpg_converted = frame2jpg(_frame.fb, 80, &_jpg_buf, &_jpg_buf_len);
            if (!_jpg_converted) {
                log_e("JPEG compression failed");
                esp_camera_fb_return(_frame.fb);
                _frame.fb = NULL;
                _jpg_buf_len = 0;
                _jpg_buf = NULL;
                return 0;
            }
        } else {
            _jpg_buf = _frame.fb->buf;
            _jpg_buf_len = _frame.fb->len;
        }

        size_t blen = 0;
        if (index) {
            blen = strlen(STREAM_BOUNDARY);
            memcpy(buffer, STREAM_BOUNDARY, blen);
            buffer += blen;
        }

        size_t hlen = sprintf((char*) buffer, STREAM_PART, JPG_CONTENT_TYPE, _jpg_buf_len);
        buffer += hlen;
        hlen = maxLen - hlen - blen;
        if (hlen > _jpg_buf_len) {
            maxLen -= hlen - _jpg_buf_len;
            hlen = _jpg_buf_len;
        }
        memcpy(buffer, _jpg_buf, hlen);
        _frame.index += hlen;
        return maxLen;
    }
    size_t available = _jpg_buf_len - _frame.index;
    if (maxLen > available) {
        maxLen = available;
    }
    memcpy(buffer, _jpg_buf + _frame.index, maxLen);
    _frame.index += maxLen;
    
    return maxLen;
}
```

Then, in your main.cpp, simply call

```
AsyncWebServerResponse* response = new AsyncJpegStreamResponse();
request->send(response);
```

closed 2024-05-27 by kRYstall9

noweare1, 2024-05-28:
Congratulations ! I will have to follow the code to see what you did. I mostly program using C.  I am not a C++ programmer so this could be a little challenging for me. The espressif IDF has a webserver  module that also supports websockets. I have been looking at that library also.