me-no-dev/ESPAsyncWebServer #1410: Compiling issues

c0debreaker, 2024-06-02:
I'm getting the errors below. If I remove `#include <ESPAsyncWebServer.h>` and other related code, the compile becomes successful. What did I do someting wrong?

```
/home/rcamara/Arduino/libraries/ESP_Async_WebServer/src/WebAuthentication.cpp: In function 'bool getMD5(uint8_t*, uint16_t, char*)':
/home/rcamara/Arduino/libraries/ESP_Async_WebServer/src/WebAuthentication.cpp:74:3: error: 'mbedtls_md5_starts_ret' was not declared in this scope; did you mean 'mbedtls_md5_starts'?
   74 |   mbedtls_md5_starts_ret(&_ctx);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_starts
/home/rcamara/Arduino/libraries/ESP_Async_WebServer/src/WebAuthentication.cpp:75:3: error: 'mbedtls_md5_update_ret' was not declared in this scope; did you mean 'mbedtls_md5_update'?
   75 |   mbedtls_md5_update_ret(&_ctx, data, len);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_update
/home/rcamara/Arduino/libraries/ESP_Async_WebServer/src/WebAuthentication.cpp:76:3: error: 'mbedtls_md5_finish_ret' was not declared in this scope; did you mean 'mbedtls_md5_finish'?
   76 |   mbedtls_md5_finish_ret(&_ctx, _buf);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_finish
/home/rcamara/Arduino/libraries/ESP_Async_WebServer/src/AsyncEventSource.cpp: In member function 'void AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*)':
/home/rcamara/Arduino/libraries/ESP_Async_WebServer/src/AsyncEventSource.cpp:188:7: error: 'ets_printf' was not declared in this scope; did you mean 'vswprintf'?
  188 |       ets_printf("ERROR: Too many messages queued\n");
      |       ^~~~~~~~~~
      |       vswprintf
/home/rcamara/Arduino/libraries/ESP_Async_WebServer/src/AsyncWebSocket.cpp: In member function 'void AsyncWebSocketClient::_queueMessage(AsyncWebSocketMessage*)':
/home/rcamara/Arduino/libraries/ESP_Async_WebServer/src/AsyncWebSocket.cpp:551:7: error: 'ets_printf' was not declared in this scope; did you mean 'vswprintf'?
  551 |       ets_printf("ERROR: Too many messages queued\n");
      |       ^~~~~~~~~~
      |       vswprintf
/home/rcamara/Arduino/libraries/ESP_Async_WebServer/src/AsyncWebSocket.cpp: In member function 'IPAddress AsyncWebSocketClient::remoteIP()':
/home/rcamara/Arduino/libraries/ESP_Async_WebServer/src/AsyncWebSocket.cpp:832:28: error: call of overloaded 'IPAddress(unsigned int)' is ambiguous
  832 |         return IPAddress(0U);
      |                            ^
In file included from /home/rcamara/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/Arduino.h:196,
                 from /home/rcamara/Arduino/libraries/ESP_Async_WebServer/src/AsyncWebSocket.cpp:21:
/home/rcamara/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/IPAddress.h:115:3: note: candidate: 'IPAddress::IPAddress(const ip_addr_t*)'
  115 |   IPAddress(const ip_addr_t *addr);
      |   ^~~~~~~~~
/home/rcamara/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/IPAddress.h:73:3: note: candidate: 'IPAddress::IPAddress(const IPAddress&)'
   73 |   IPAddress(const IPAddress &address);
      |   ^~~~~~~~~
/home/rcamara/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/IPAddress.h:72:3: note: candidate: 'IPAddress::IPAddress(const char*)'
   72 |   IPAddress(const char *address);
      |   ^~~~~~~~~
/home/rcamara/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/IPAddress.h:69:3: note: candidate: 'IPAddress::IPAddress(const uint8_t*)'
   69 |   IPAddress(const uint8_t *address);
      |   ^~~~~~~~~
/home/rcamara/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/IPAddress.h:67:3: note: candidate: 'IPAddress::IPAddress(uint32_t)'
   67 |   IPAddress(uint32_t address);
      |   ^~~~~~~~~
/home/rcamara/Arduino/libraries/ESP_Async_WebServer/src/AsyncWebSocket.cpp: In constructor 'AsyncWebSocketResponse::AsyncWebSocketResponse(const String&, AsyncWebSocket*)':
/home/rcamara/Arduino/libraries/ESP_Async_WebServer/src/AsyncWebSocket.cpp:1262:3: error: 'mbedtls_sha1_starts_ret' was not declared in this scope; did you mean 'mbedtls_sha1_starts'?
 1262 |   mbedtls_sha1_starts_ret(&ctx);
      |   ^~~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_sha1_starts
/home/rcamara/Arduino/libraries/ESP_Async_WebServer/src/AsyncWebSocket.cpp:1263:3: error: 'mbedtls_sha1_update_ret' was not declared in this scope; did you mean 'mbedtls_sha1_update'?
 1263 |   mbedtls_sha1_update_ret(&ctx, (const unsigned char*)key.c_str(), key.length());
      |   ^~~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_sha1_update
/home/rcamara/Arduino/libraries/ESP_Async_WebServer/src/AsyncWebSocket.cpp:1264:3: error: 'mbedtls_sha1_finish_ret' was not declared in this scope; did you mean 'mbedtls_sha1_finish'?
 1264 |   mbedtls_sha1_finish_ret(&ctx, hash);
      |   ^~~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_sha1_finish
Multiple libraries were found for "WiFi.h"
  Used: /home/rcamara/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/WiFi
  Not used: /home/rcamara/Arduino/libraries/WiFiNINA
exit status 1

Compilation error: exit status 1
```

and my source code is
```
#include <Arduino.h>
#include <WiFi.h>
#include <LittleFS.h>
#include <ESPAsyncWebServer.h>

// Replace with your network credentials
const char* ssid = "******";
const char* password = "******";

AsyncWebServer server(80);

void setup() {
  // Start Serial for debug output
  Serial.begin(115200);

  // Initialize LittleFS
  if (!LittleFS.begin()) {
    Serial.println("An Error has occurred while mounting LittleFS");
    return;
  }

  // Connect to Wi-Fi
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi...");
  }
  Serial.println("Connected to WiFi");

  // Serve static files from LittleFS
  server.serveStatic("/", LittleFS, "/").setDefaultFile("index.html");

  // Handle 404 (Not Found)
  server.onNotFound([](AsyncWebServerRequest *request) {
    request->send(404, "text/plain", "Not found");
  });

  // Start server
  server.begin();
}

void loop() {
  // Nothing needed here
}

```

lbernstone, 2024-06-02:
This needs guardrails for when to include, but here are the necessary includes
```
diff --git a/src/AsyncEventSource.cpp b/src/AsyncEventSource.cpp
index f2914df..230e3f6 100644
--- a/src/AsyncEventSource.cpp
+++ b/src/AsyncEventSource.cpp
@@ -19,6 +19,7 @@
 */
 #include "Arduino.h"
 #include "AsyncEventSource.h"
+#include "rom/ets_sys.h"
 
 static String generateEventMessage(const char *message, const char *event, uint32_t id, uint32_t reconnect){
   String ev = "";
diff --git a/src/AsyncWebSocket.cpp b/src/AsyncWebSocket.cpp
index f76f2fc..ef94775 100644
--- a/src/AsyncWebSocket.cpp
+++ b/src/AsyncWebSocket.cpp
@@ -28,6 +28,7 @@
 #else
 #include <Hash.h>
 #endif
+#include "mbedtls/compat-2.x.h"
 
 #define MAX_PRINTF_LEN 64
 
@@ -829,7 +830,7 @@ void AsyncWebSocketClient::binary(AsyncWebSocketMessageBuffer * buffer)
 
 IPAddress AsyncWebSocketClient::remoteIP() {
     if(!_client) {
-        return IPAddress(0U);
+        return IPAddress(0,0,0,0);
     }
     return _client->remoteIP();
 }
diff --git a/src/WebAuthentication.cpp b/src/WebAuthentication.cpp
index 45246a1..01fb567 100644
--- a/src/WebAuthentication.cpp
+++ b/src/WebAuthentication.cpp
@@ -22,6 +22,7 @@
 #include <libb64/cencode.h>
 #ifdef ESP32
 #include "mbedtls/md5.h"
+#include "mbedtls/compat-2.x.h"
 #else
 #include "md5.h"
 #endif
```

c0debreaker, 2024-06-02:
> This needs guardrails for when to include, but here are the necessary includes
>
> ```
> diff --git a/src/AsyncEventSource.cpp b/src/AsyncEventSource.cpp
> index f2914df..230e3f6 100644
> --- [...]

Hi lbernstone,

I see there are 3 files that needs to be updated. I can update it manually. However, I like to learn how to update it from your given code. What tool should I use for that? It looks similar to some diffs I've seen very long time ago using `patch` command. Thanks a lot!

lbernstone, 2024-06-02:
That was a git diff.  It was updated with my own 2 hands :) I just looked at what was missing, grep'd for those functions in the arduino-esp32/tools/esp32-arduino-libs/esp32/include folder, and included those files in the correct locations.  To make a PR, I need to figure out a way to only include those headers if the version is >3.0.0.

c0debreaker, 2024-06-03:
Oh got it. I knew I will have issues when Arduino popped up something on the lower right corner. `Do you want to update to 3.0.0?` I said yes. Few hours later, I encountered the issues above.

Alrighty, I'll make changes to my local. Thank you.

jimdgit, 2024-06-03:
I am having the same issue, but I am sorry I do not understand the fix.
Could you please explain what fixes I need to make?
Thank You

capc0307, 2024-06-13:
> That was a git diff. It was updated with my own 2 hands :) I just looked at what was missing, grep'd for those functions in the arduino-esp32/tools/esp32-arduino-libs/esp32/include folder, and [...]

I found this: 

```
#ifdef ESP_ARDUINO_VERSION_MAJOR
            #if ESP_ARDUINO_VERSION >= ESP_ARDUINO_VERSION_VAL(3, 0, 0)
                                                  // Code for version 3.x

                   #else
                                                 // Code for version 2.x
                   #ifdef ESP32

                   #else
 

                    #endif
                 #endif
                #endif

```
on https://github.com/h2zero/NimBLE-Arduino/issues/676#issuecomment-2160799917, from a comment made by [ednieuw](https://github.com/ednieuw). ednieuw says they got this piece of code from the Espressif migration 2.x -> 3.x document. Supposedly this piece of code makes a library compatible for both ESP32 versions. I hope this helps.

joyfullservice, 2024-06-14:
> If I comment out line 551 then it all compiles.

Just an observation, but in `AsyncWebSocket.cpp` the `ets_printf()` function is used 7 times, but 6 of these are commented out. The one that triggers the compile error is the only one that is not commented out. Perhaps that one got missed when the debug lines were added and commented out six years ago.

I can confirm that after making the changes above and commenting out this line, it compiles and seems to work fine on an ESP32 dev board.

kalwinskidawid, 2024-08-14:
Is this already fixed?
I am trying to build a project with this library and I have the same problem.
In particular, there is a problem when I execute request->send().

seandepagnier, 2024-08-28:
i changed it to printf... dont do that!

so maybe ::printf instead of ets_printf?

aranFish, 2024-09-06:
I also encountered these same issues, and also fixed the by making the suggested modifications above.
I'm getting one last compilation error that is not mentioned above, however:

`Linking everything together...
/Users/aran/Library/Arduino15/packages/esp32/tools/esp-x32/2405/bin/xtensa-esp32-elf-g++ @/Users/aran/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-466a392a/esp32/flags/ld_flags @/Users/aran/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-466a392a/esp32/flags/ld_scripts -Wl,--Map=/private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/BlinkRGB-esp32c6.ino.map -L/Users/aran/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-466a392a/esp32/lib -L/Users/aran/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-466a392a/esp32/ld -L/Users/aran/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-466a392a/esp32/dio_qspi -Wl,--wrap=esp_panic_handler -Wl,--start-group /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/sketch/BlinkRGB-esp32c6.ino.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/WiFi/AP.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/WiFi/STA.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/WiFi/WiFi.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/WiFi/WiFiAP.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/WiFi/WiFiGeneric.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/WiFi/WiFiMulti.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/WiFi/WiFiSTA.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/WiFi/WiFiScan.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/Network/NetworkClient.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/Network/NetworkEvents.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/Network/NetworkInterface.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/Network/NetworkManager.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/Network/NetworkServer.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/Network/NetworkUdp.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/AsyncTCP/AsyncTCP.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/ESPAsyncWebServer/AsyncEventSource.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/ESPAsyncWebServer/AsyncWebSocket.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/ESPAsyncWebServer/WebAuthentication.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/ESPAsyncWebServer/WebHandlers.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/ESPAsyncWebServer/WebRequest.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/ESPAsyncWebServer/WebResponses.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/ESPAsyncWebServer/WebServer.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/FS/FS.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/FS/vfs_api.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/LittleFS/LittleFS.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/Arduino_JSON/JSON.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/Arduino_JSON/JSONVar.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/Arduino_JSON/cjson/cJSON.c.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/core/variant.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/cores/9dc0c0ccb082ebf96739fc43da6b9c53/core.a @/Users/aran/Library/Arduino15/packages/esp32/tools/es [... cut ...] BlinkRGB-esp32c6.ino.elf
/Users/aran/Library/Arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/95EF2D842E5576E281114D8C428A7F6E/libraries/ESPAsyncWebServer/AsyncWebSocket.cpp.o:(.literal._ZN17AsyncWebLockGuardC5ERK12AsyncWebLock[_ZN17AsyncWebLockGuardC5ERK12AsyncWebLock]+0x0): undefined reference to `pxCurrentTCB'
collect2: error: ld returned 1 exit status
`

aranFish, 2024-09-06:
I've encountered the same errors as the original poster, and managed to fix them by following @lbernstone 's recommendations for extra #include statements, as well as @jimdgit 's recommendation to comment out line #551 in AsyncWebSocket.cpp.

However, I'm still getting one more error. I think it's because my code uses 

Here's the compilation error message:
```
/Users/aran/Library/Arduino15/packages/esp32/tools/esp-x32/2405/bin/xtensa-esp32-elf-g++ @/Users/aran/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-466a392a/esp32/flags/ld_flags @/Users/aran/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-466a392a/esp32/flags/ld_scripts -Wl,--Map=/private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/sketch_sep6b.ino.map -L/Users/aran/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-466a392a/esp32/lib -L/Users/aran/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-466a392a/esp32/ld -L/Users/aran/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-466a392a/esp32/dio_qspi -Wl,--wrap=esp_panic_handler -Wl,--start-group /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/sketch/sketch_sep6b.ino.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/WiFi/AP.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/WiFi/STA.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/WiFi/WiFi.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/WiFi/WiFiAP.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/WiFi/WiFiGeneric.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/WiFi/WiFiMulti.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/WiFi/WiFiSTA.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/WiFi/WiFiScan.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/Network/NetworkClient.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/Network/NetworkEvents.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/Network/NetworkInterface.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/Network/NetworkManager.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/Network/NetworkServer.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/Network/NetworkUdp.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/ESPAsyncWebServer/AsyncEventSource.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/ESPAsyncWebServer/AsyncWebSocket.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/ESPAsyncWebServer/WebAuthentication.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/ESPAsyncWebServer/WebHandlers.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/ESPAsyncWebServer/WebRequest.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/ESPAsyncWebServer/WebResponses.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/ESPAsyncWebServer/WebServer.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/FS/FS.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/FS/vfs_api.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/AsyncTCP/AsyncTCP.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/libraries/LittleFS/LittleFS.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/core/variant.cpp.o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/cores/9dc0c0ccb082ebf96739fc43da6b9c53/core.a @/Users/aran/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-466a392a/esp32/flags/ld_libs -Wl,--end-group -Wl,-EL -o /private/var/folders/0p/vzf6vcfj7_z0hqd8j7ksbymh0000gn/T/arduino/sketches/A9D9498DC5653A959082DB29939625EC/sketch_sep6b.ino.elf
[... cut ...]
```

But I'm not sure what to try next to debug this. Please could you give me some help for what to try next?

Arduino Version: 2.3.2

Using library WiFi at version 3.1.0
Using library Networking at version 3.1.0
Using library ESP Async WebServer at version 1.2.4
Using library FS at version 3.1.0
Using library AsyncTCP at version 1.1.4
Using library LittleFS at version 3.1.0

Mac OS 13.5
Using Adafruit Huzzah32 Feather v2 (as [here](https://learn.adafruit.com/adafruit-esp32-feather-v2/overview))

aranFish, 2024-09-07:
OK I've made the lock() function in AsyncWebSynchronization.h always return false, by commenting out other lines inside that function. My very basic code is now working. I'll keep an eye out for what I've broken as I build up the complexity of the firmware again.

MOUMOUDOU, 2024-11-30:
A good fix I found and tried with success is to modify the file AsyncWebSynchronization.h using the following steps :

1. Add after the `#ifdef ESP32` at line 9 : `#include "esp_private/freertos_debug.h"`.

2. Modify line 29 with: `void *_pxCurrentTCB = pvTaskGetCurrentTCBForCore(esp_cpu_get_core_id());`.

3. Replace subsequent references of `pxCurrentTCB` with `_pxCurrentTCB` (at line 30 and 32).

per1234, 2024-12-17:
This has been fixed by https://github.com/me-no-dev/ESPAsyncWebServer/commit/17a8d442169f3ca8496b20ebd24bbc06f55d4a62

@c0debreaker please install the latest version of the library from this GitHub repository (make sure you don't install one of the forks which haven't pulled down the fix), then check to see if the bug is fixed. If so, you can close the issue by clicking the "**Close issue**" button here on the issue web page.

PureTek-Innovations, 2025-01-02:
@per1234 
I'm using version 1.2.4 straight from this repository with Arduino core 3.1.0 and am still getting the pxCurrentTCB issue.
It doesn't appear to have been fixed.
Thanks

misterjtc, 2025-01-03:
I am using version 1.2.9 and esp32 3.1.0 and also getting the issue.

per1234, 2025-01-03:
> still getting the pxCurrentTCB issue

@PureTek-Innovations this issue is not about `pxCurrentTCB`, and my comment was not about `pxCurrentTCB`. The problem with `pxCurrentTCB` is tracked in a separate issue: https://github.com/me-no-dev/ESPAsyncWebServer/issues/1442. So it is not productive to discuss it here.

per1234, 2025-01-03:
>I am using version 1.2.9 

@misterjtc there is no such version so you must be using one of the forks of the repository. Uninstall that version and then install the library from this repository. You can't install this library via the Arduino Library Manager (https://github.com/me-no-dev/ESPAsyncWebServer/issues/1433). If you are using Arduino IDE, you can install it by the following procedure:

1. Click the following link to open the library's GitHub repository homepage in your web browser:
   <https://github.com/me-no-dev/ESPAsyncWebServer>
1. Click the "**Code ▾**" button you see on that page.
1. Select **Download ZIP** from the menu.
   A download of the ZIP file of the library will start.
1. Wait for the download to finish.
1. Select **Sketch > Include library > Add .ZIP Library** from the Arduino IDE menus.
   The "**Select the zip file containing the library you'd like to add**" dialog will open.
1. Select the downloaded file from the dialog.
1. Click the "**Open**" button.
   The dialog will close.
1. Wait for the installation process to finish, as indicated by a notification at the bottom right corner of the Arduino IDE window:
   > **ⓘ Successfully installed library from ...**