me-no-dev/ESPAsyncWebServer #1413: small ddos (press F5) crash: async_tcp ....watchdog got triggered

Red-Owl, 2024-06-06:
multiple request on file'S cause: 

E (47645) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (47645) task_wdt:  - async_tcp (CPU 0/1)
E (47645) task_wdt: Tasks currently running:
E (47645) task_wdt: CPU 0: IDLE
E (47645) task_wdt: CPU 1: async_tcp
E (47645) task_wdt: Aborting.

the server restart. server crash in random where normal webpage read or update websoket data

Red-Owl, 2024-07-09:
server.serveStatic("/", SD, "/").setCacheControl("max-age=3600");   

<-- multiple (many) request cause server panic in async tcp
if remove this function and send muliple request server not crash

StudBoy646, 2024-08-08:
Can you please tell me in which file this function is present ?