me-no-dev/ESPAsyncWebServer #1414: Compilation Error for Newer ESP32

bitscoper, 2024-06-08:
```
Compiling library "ESPAsyncWebServer-master"
/home/bitscoper/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/bin/xtensa-esp32-elf-g++ -MMD -c @/home/bitscoper/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/flags/cpp_flags -Wall -Wextra -Os -DF_CPU=240000000L -DARDUINO=10819 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DARDUINO_PARTITION_huge_app "-DARDUINO_HOST_OS=\"linux\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32:JTAGAdapter=default,PSRAM=disabled,PartitionScheme=huge_app,CPUFreq=240,FlashMode=dio,FlashFreq=40,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=warn,EraseFlash=none,ZigbeeMode=default\"" -DESP32 -DCORE_DEBUG_LEVEL=2 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 @/home/bitscoper/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/flags/defines "-I/home/bitscoper/Software Laboratory/Bitscoper_IoT/Master" -iprefix /home/bitscoper/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/include/ @/home/bitscoper/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/flags/includes -I/home/bitscoper/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/dio_qspi/include -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32 -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/variants/esp32 -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/FS/src -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/SPIFFS/src -I/home/bitscoper/Arduino/libraries/LiquidCrystal_I2C -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/Wire/src -I/home/bitscoper/Arduino/libraries/ESP32Servo/src -I/home/bitscoper/Arduino/libraries/rdm6300-master/src -I/home/bitscoper/Arduino/libraries/MPU9250_WE/src -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/SPI/src -I/home/bitscoper/Arduino/libraries/BH1750/src -I/home/bitscoper/Arduino/libraries/Grove_-_Barometer_Sensor_BME280 -I/home/bitscoper/Arduino/libraries/TinyGPSPlus-ESP32/src -I/home/bitscoper/Arduino/libraries/Lewis/src -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/WiFi/src -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/Network/src -I/home/bitscoper/Arduino/libraries/AsyncTCP-master/src -I/home/bitscoper/Arduino/libraries/ESPAsyncWebServer-master/src -I/home/bitscoper/Arduino/libraries/ArduinoJson/src @/tmp/arduino_build_14758/build_opt.h @/tmp/arduino_build_14758/file_opts /home/bitscoper/Arduino/libraries/ESPAsyncWebServer-master/src/WebAuthentication.cpp -o /tmp/arduino_build_14758/libraries/ESPAsyncWebServer-master/WebAuthentication.cpp.o
/home/bitscoper/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/bin/xtensa-esp32-elf-g++ -MMD -c @/home/bitscoper/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/flags/cpp_flags -Wall -Wextra -Os -DF_CPU=240000000L -DARDUINO=10819 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DARDUINO_PARTITION_huge_app "-DARDUINO_HOST_OS=\"linux\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32:JTAGAdapter=default,PSRAM=disabled,PartitionScheme=huge_app,CPUFreq=240,FlashMode=dio,FlashFreq=40,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=warn,EraseFlash=none,ZigbeeMode=default\"" -DESP32 -DCORE_DEBUG_LEVEL=2 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 @/home/bitscoper/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/flags/defines "-I/home/bitscoper/Software Laboratory/Bitscoper_IoT/Master" -iprefix /home/bitscoper/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/include/ @/home/bitscoper/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/flags/includes -I/home/bitscoper/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/dio_qspi/include -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32 -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/variants/esp32 -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/FS/src -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/SPIFFS/src -I/home/bitscoper/Arduino/libraries/LiquidCrystal_I2C -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/Wire/src -I/home/bitscoper/Arduino/libraries/ESP32Servo/src -I/home/bitscoper/Arduino/libraries/rdm6300-master/src -I/home/bitscoper/Arduino/libraries/MPU9250_WE/src -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/SPI/src -I/home/bitscoper/Arduino/libraries/BH1750/src -I/home/bitscoper/Arduino/libraries/Grove_-_Barometer_Sensor_BME280 -I/home/bitscoper/Arduino/libraries/TinyGPSPlus-ESP32/src -I/home/bitscoper/Arduino/libraries/Lewis/src -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/WiFi/src -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/Network/src -I/home/bitscoper/Arduino/libraries/AsyncTCP-master/src -I/home/bitscoper/Arduino/libraries/ESPAsyncWebServer-master/src -I/home/bitscoper/Arduino/libraries/ArduinoJson/src @/tmp/arduino_build_14758/build_opt.h @/tmp/arduino_build_14758/file_opts /home/bitscoper/Arduino/libraries/ESPAsyncWebServer-master/src/AsyncEventSource.cpp -o /tmp/arduino_build_14758/libraries/ESPAsyncWebServer-master/AsyncEventSource.cpp.o
/home/bitscoper/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/bin/xtensa-esp32-elf-g++ -MMD -c @/home/bitscoper/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/flags/cpp_flags -Wall -Wextra -Os -DF_CPU=240000000L -DARDUINO=10819 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DARDUINO_PARTITION_huge_app "-DARDUINO_HOST_OS=\"linux\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32:JTAGAdapter=default,PSRAM=disabled,PartitionScheme=huge_app,CPUFreq=240,FlashMode=dio,FlashFreq=40,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=warn,EraseFlash=none,ZigbeeMode=default\"" -DESP32 -DCORE_DEBUG_LEVEL=2 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 @/home/bitscoper/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/flags/defines "-I/home/bitscoper/Software Laboratory/Bitscoper_IoT/Master" -iprefix /home/bitscoper/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/include/ @/home/bitscoper/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/flags/includes -I/home/bitscoper/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/dio_qspi/include -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32 -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/variants/esp32 -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/FS/src -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/SPIFFS/src -I/home/bitscoper/Arduino/libraries/LiquidCrystal_I2C -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/Wire/src -I/home/bitscoper/Arduino/libraries/ESP32Servo/src -I/home/bitscoper/Arduino/libraries/rdm6300-master/src -I/home/bitscoper/Arduino/libraries/MPU9250_WE/src -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/SPI/src -I/home/bitscoper/Arduino/libraries/BH1750/src -I/home/bitscoper/Arduino/libraries/Grove_-_Barometer_Sensor_BME280 -I/home/bitscoper/Arduino/libraries/TinyGPSPlus-ESP32/src -I/home/bitscoper/Arduino/libraries/Lewis/src -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/WiFi/src -I/home/bitscoper/.arduino15/packages/esp32/hardware/esp32/3.0.0/libraries/Network/src -I/home/bitscoper/Arduino/libraries/AsyncTCP-master/src -I/home/bitscoper/Arduino/libraries/ESPAsyncWebServer-master/src -I/home/bitscoper/Arduino/libraries/ArduinoJson/src @/tmp/arduino_build_14758/build_opt.h @/tmp/arduino_build_14758/file_opts /home/bitscoper/Arduino/libraries/ESPAsyncWebServer [... cut ...]
/home/bitscoper/Arduino/libraries/ESPAsyncWebServer-master/src/AsyncWebSocket.cpp:1263:3: error: 'mbedtls_sha1_update_ret' was not declared in this scope; did you mean 'mbedtls_sha1_update'?
 1263 |   mbedtls_sha1_update_ret(&ctx, (const unsigned char*)key.c_str(), key.length());
      |   ^~~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_sha1_update
/home/bitscoper/Arduino/libraries/ESPAsyncWebServer-master/src/AsyncWebSocket.cpp:1264:3: error: 'mbedtls_sha1_finish_ret' was not declared in this scope; did you mean 'mbedtls_sha1_finish'?
 1264 |   mbedtls_sha1_finish_ret(&ctx, hash);
      |   ^~~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_sha1_finish
```

mlthurtell, 2024-06-08:
Can confirm I am also getting this whilst trying to build on 5.1.4

me-no-dev, 2024-06-08:
this has been fixed. just pull the latest source

closed 2024-06-08 by me-no-dev