me-no-dev/ESPAsyncWebServer #1419: Problem resolved ?   "Arduino\libraries\ESPAsyncWebSrv\src\WebAuthentication.cpp:74:3: error: 'mbedtls_md5_starts_ret' was not declared in this scope; did you mean 'mbedtls_md5_starts'?"

A2028, 2024-06-28:
As the ESP32 Boards Package updated to V3 in Arduino IDE, many issues or compilation errors come out, any updated version of ESPAsyncWebServer resolve the above problem?

me-no-dev, 2024-07-01:
update your ESPAsyncWebServer library. It has been fixed to compile and work with v3

iammatis, 2024-07-03:
I don't think it's been fixed yet

```shell
/Arduino/libraries/ESPAsyncWebServer/src/WebAuthentication.cpp: In function 'bool getMD5(uint8_t*, uint16_t, char*)':
/Arduino/libraries/ESPAsyncWebServer/src/WebAuthentication.cpp:74:3: error: 'mbedtls_md5_starts_ret' was not declared in this scope; did you mean 'mbedtls_md5_starts'?
   74 |   mbedtls_md5_starts_ret(&_ctx);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_starts
/Arduino/libraries/ESPAsyncWebServer/src/WebAuthentication.cpp:75:3: error: 'mbedtls_md5_update_ret' was not declared in this scope; did you mean 'mbedtls_md5_update'?
   75 |   mbedtls_md5_update_ret(&_ctx, data, len);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_update
/Arduino/libraries/ESPAsyncWebServer/src/WebAuthentication.cpp:76:3: error: 'mbedtls_md5_finish_ret' was not declared in this scope; did you mean 'mbedtls_md5_finish'?
   76 |   mbedtls_md5_finish_ret(&_ctx, _buf);
      |   ^~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_md5_finish
/Arduino/libraries/ESPAsyncWebServer/src/AsyncEventSource.cpp: In member function 'void AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*)':
/Arduino/libraries/ESPAsyncWebServer/src/AsyncEventSource.cpp:189:7: error: 'ets_printf' was not declared in this scope; did you mean 'vswprintf'?
  189 |       ets_printf("ERROR: Too many messages queued\n");
      |       ^~~~~~~~~~
      |       vswprintf
/Arduino/libraries/ESPAsyncWebServer/src/AsyncWebSocket.cpp: In member function 'void AsyncWebSocketClient::_queueMessage(AsyncWebSocketMessage*)':
/Arduino/libraries/ESPAsyncWebServer/src/AsyncWebSocket.cpp:549:7: error: 'ets_printf' was not declared in this scope; did you mean 'vswprintf'?
  549 |       ets_printf("ERROR: Too many messages queued\n");
      |       ^~~~~~~~~~
      |       vswprintf

Error during build: exit status 1
```

Versions:
```
ESPAsyncWebServer: 3.1.0
esp32: 3.0.2
```

Ollie9T7, 2024-07-03:
Still not fixed

al-TideLight, 2024-07-05:
No fixed for me either, latest versions:
ESPAsyncWebServer: 3.1.0
esp32: 3.0.2

lizardb0y, 2024-07-06:
> No fixed for me either, latest versions: ESPAsyncWebServer: 3.1.0 esp32: 3.0.2

The library available through the Arduino Library Manager (ESPAsyncWebServer: 3.1.0 ) is a different fork maintained by someone else. At the time I write this that fork is still failing to build. 

This fork, maintained by me-no-dev, does not have those errors.

You will need to remove the ESPAsyncWebServer library then install the library from this repository manually.

al-TideLight, 2024-07-06:
Ok ideal thanks I'll try that this week and report back

A2028, 2024-07-07:
Well, the version is ESPAsyncWebSrv Version 1.2.7

It still has error ....
[image]

A2028, 2024-07-07:
The compilation was successful at ESP32 V2 and no AsyncWebServer installed, just installed the AsyncWebSrv (up to 1.2.7)

NOT successful at ESP32 V3.x

ednieuw, 2024-07-07:
I solved your error in the library from this fork
https://github.com/lacamera/ESPAsyncWebServer/pull/2
That will probably also work here 

Changed in Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp
at line 75,76,77
//-----------------
#ifdef ESP_ARDUINO_VERSION_MAJOR
#if ESP_ARDUINO_VERSION >= ESP_ARDUINO_VERSION_VAL(3, 0, 0)
// Code for version 3.x
mbedtls_md5_init(&_ctx);
mbedtls_md5_starts(&_ctx);
mbedtls_md5_update(&_ctx, data, len);
mbedtls_md5_finish(&_ctx, _buf);
#else
// Code for version 2.x
#ifdef ESP32
mbedtls_md5_init(&_ctx);
mbedtls_md5_starts_ret(&_ctx);
mbedtls_md5_update_ret(&_ctx, data, len);
mbedtls_md5_finish_ret(&_ctx, _buf);
#else
MD5Init(&_ctx);
MD5Update(&_ctx, data, len);
MD5Final(_buf, &_ctx);
#endif
#endif
#endif
//--------------------
in AsyncEventSource.cpp changed at line 189 ets_printf --> log_e for V3
in AsyncWebSocket.cpp changed at line 549 ets_printf --> log_e for V3
#ifdef ESP_ARDUINO_VERSION_MAJOR
#if ESP_ARDUINO_VERSION >= ESP_ARDUINO_VERSION_VAL(3, 0, 0)
// Code for version 3.x
log_e("ERROR: Too many messages queued\n");
#else
// Code for version 2.x
ets_printf("ERROR: Too many messages queued\n");
#endif
#endif

A2028, 2024-07-08:
👍🏻

closed 2024-07-08 by A2028