me-no-dev/ESPAsyncWebServer #1422: IDF 5 Coexistence support

jcfain, 2024-07-12:
Hello,this may not be an issue but I just wanted to make sure I am not doing something incorrect.

I am trying to use this web server along with NimBLE in arduino-esp32 using a platform branch 
https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5 which is based off ESPIDF 5.1.4.

I have gotten it to work with another web server (PsychicHttp) but it didn't seem very stable at this point.
This server works great but as soon as I throw NimBLE into the mix it doesn't at all.

Is this supported yet? Do I need to change anything?
If not, am I safe to assume support will be coming eventually?

Thanks!

me-no-dev, 2024-07-15:
Current master of the web server supports Arduino v3 on IDF 5.1.4

jcfain, 2024-08-08:
> Current master of the web server supports Arduino v3 on IDF 5.1.4

Yes, sorry I wasn't clear. The Master branch DOES work as long as I don't start the NimBLE service

jcfain, 2024-08-09:
So after further investigation, I discovered the web server doesn't work even if I don't initiate the BLE server but leave the code in for a software toggle. Web server works fine if the BLE library is left out of the build.
I'm thinking this maybe a memory issue. 
I'm using the Do-it Devkit V1 with wroom32D module.

jcfain, 2024-08-25:
Using the following pio config:
```
platform = espressif32@6.7.0
framework = arduino

platform_packages =
	platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.4
	platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.4/esp32-arduino-libs-3.0.4.zip
```

I can send data over both UDP using AsyncUDP and NimBLE and these two seem to coexist just fine.
When I try to access my webserver though the page never shows up. Even if I disable bt with
`esp_bt_controller_deinit()`
I have my lib-deps pointed to the latest git release of ESPAsyncWebServer.

jcfain, 2024-08-25:
Ok, I think I have come to the conclusion it IS a memory issue. I discovered esp_bt_controller_mem_release and when the ble is not initialized, the webserver works again.

closed 2024-08-25 by jcfain