me-no-dev/ESPAsyncWebServer #1424: Cant set Table or Cell width % because of template processing

MarksWizard, 2024-07-20:
Trying to add the following HTML breaks because it picks up the width % as a template.  Is there any way to make it keep the % as text like I would do with <font size=\"4\">?  For now, my work around is to pass the whole style as a template.

<style> 
html {font-family: Helvetica; display: inline; margin: auto 0; text-align: center;}
table {width: 100%; text-align: center;} 
td {width: 50%; text-align: center; padding: 0 20px;}
</style>

MarksWizard, 2024-07-20:
I found a solution in issue[ #1249 ](https://github.com/me-no-dev/ESPAsyncWebServer/issues/1249).  Adding an extra % fixed the issue in my case.

`<style> 
html {font-family: Helvetica; display: inline; margin: auto 0; text-align: center;}
table {width: 100%%; text-align: center;} 
td {width: 50%%; text-align: center; padding: 0 20px;}
</style>`

munkerench, 2024-07-25:
I would suggest making the delimiter a char 3 and allowing the value to be passed in as a param to the processor.  One character really limits the capability.  Much more flexible.  if the strings or too flexible are a concern, store them in an enum, or limit them to some set of chars like ! @ # $ % ^ & *.

georgepapas123, 2024-10-17:
I second [munkerench](https://github.com/munkerench) suggestion. 
My problematic html was width: 100%; height: 100%;
which was picked up as parameter "; height: 100", it took me a while to realize why the page did not behave as expected.
My solution was to use instead width: 100vw; height: 100vh; but specifying the delimiter would solve this type of problem once and for all.
Please consider it.