me-no-dev/ESPAsyncWebServer #1431: Simple example of read/write to SPIFFS

y-nk, 2024-09-07:
Hello,

I'm looking to build something which looks trivial, but never succeed.

```c
#include <WiFi.h>
#include <ESPAsyncWebServer.h>

const char html[] = R"__(
<body>
  <form method="POST" action="/">
    <textarea name="content" value="%VALUE%"></textarea>
    <button type="submit">submit</button>
  </form>
  <style>
  </style>
</body>
)__";

String getFileContent() {
  File file = SPIFFS.open("/data", FILE_READ);
  
  if (!file) {
    Serial.println("Failed to open file for reading");
    return "";
  }
  
  String data = file.readString();
  file.close();

  return data;
}

void setFileContent(String content) {
  File file = SPIFFS.open("/data", FILE_WRITE);
  
  if (!file) {
    Serial.println("Failed to open file for writing");
    return;
  }
   
  while(!file.availableForWrite());
 
  int bytes = 0;
  do {
    bytes = file.println(content);
    Serial.printf("written content -> %s", bytes);
  } while (bytes == 0);

  file.close();
}

AsyncWebServer* server;

void setup() {
  Serial.begin(115200);
  while (!Serial);

  if (!SPIFFS.begin(true)) {
    Serial.println("SPIFFS Error");
    return;
  }

  WiFi.mode(WIFI_STA);
  WiFi.setHostname("ESP32");
  WiFi.begin("ESP32", "testtest");

  // webserver setup
  server = new AsyncWebServer(80);

  server->on("/", HTTP_GET, [](AsyncWebServerRequest *req) {
    String data = String(html);
    data.replace("%VALUE%", getFileContent());

    req->send(200, "text/html", data);
  });

  server->on("/", HTTP_POST, [](AsyncWebServerRequest *req) {
    int params = req->params();

    AsyncWebParameter* p;

    for (int i = 0; i < params; i++) {
      p = req->getParam(i);
      if (p->name() == "content") break;
    }

    setFileContent(p->value());
    req->redirect("/");
  });

  server->onNotFound([](AsyncWebServerRequest *req) {
    req->send(404, "text/plain", "Not Found");
  });

  server->begin();
}

void loop() {
}
```

When running this:

1. AP works
2. Server serves GET /
3. Form submits to POST /
4. Panic error comes in about the moment of the write.

I wonder why such a small example is not working, especially that i submit about 10 bytes of text in the textarea.

stevesch, 2024-09-08:
Just looking briefly, but maybe the

while(!file.availableForWrite());

... Maybe yield in that loop.  I'd avoid busy wait altogether if possible.

y-nk, 2024-09-09:
@stevesch i'm sorry not to have updated sooner. I've spent good of time trying out and finding my issue. I isolated out the web server and found that it was a SPIFFS lib issue instead (It went to use ESP32 Exception decoder with a v1 of Arduino)

TLDR: there's an issue with using `String` class in `file.println(content);` it looks like not working at all.

[If you're interested into why/how.](https://www.reddit.com/r/esp32/comments/1fbe7ea/writing_on_spiffs_causes_guru_meditation_error/)

For people passing here with a hope for a fix:

```c
    String value = param->value();
    
    char str[value.length() + 1];
    value.toCharArray(str, value.length() + 1);

    setFileContent(str);
```

with:

```c
void setFileContent(const char* content);
```

Closing because it's not related to the server at all. Sorry for wasting your time.

closed 2024-09-09 by y-nk