me-no-dev/ESPAsyncWebServer #1433: Library accessible from the Arduino IDE Lib Manager

Zeppelin500, 2024-09-16:
Hello, 

It would be great if the library was available in the Arduino IDE library manager.
Sure, i can fork the library and make it accessible from the manager, but the original would be much better.

There is already a fork in the manager, but it is so old that it no longer works.
Thank you!

s-light, 2024-12-17:
@Zeppelin500 you do not need to fork the library to get it into the manager - 
as long as the library fullfills the requiremetns:
- [requirements for successful add...](https://github.com/arduino/library-registry/blob/main/FAQ.md#what-are-the-requirements-for-a-library-to-be-added-to-library-manager))*
- https://github.com/arduino/library-registry/blob/main/README.md#adding-a-library-to-library-manager

i have done this [myself for this library](https://github.com/codingABI/DFR0534/issues/1)

and you can check if the requirements are fulfilled before you do the process..
currently missing is at least  a release / tag..

so best thing: one of the users checks if the requiremetns are allready fine.
and if not creates a pr with the needed minimal changes..
and clear wisches ;-)
so that me-no-dev has the least amount of work to do to get it to a fine state - 
than someone else can doe the pullrequest for the library-list..

maybe this way we have a chance... ;-)

s-light, 2024-12-17:
depends on https://github.com/me-no-dev/ESPAsyncWebServer/issues/1427