me-no-dev/ESPAsyncWebServer #1435: Template processing error with IOS18

hakkapet, 2024-09-25:
Hi, just noticed that some reason when Iphone is updated to **IOS18**. 
**Template processing** wont work anymore in Safari browser: It just gives an "**cannot parse response**" error to Safari.  So i this is Safari/IOS18 error, but it would be nice to know why this wont work anymore. It is crucial functionality almost everywhere in web applications. Other browsers (Chrome) what i was trying to use works fine. Some reason return page from **AwsTemplateProcessor** is interpreted as faulty in Safari.

Example this wont work anymore:
```
server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
        const char * html = "<p>Temperature: %PLACEHOLDER_TEMPERATURE% ºC</p><p>Humidity: %PLACEHOLDER_HUMIDITY% %</p>";
          request->send_P(200, "text/html", html, processor); 
});
```
or this wont work:
```
    server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
          request->send(LittleFS, "/index.html", "text/html;", false, processor);
    });
```
Without processor it gives page without problem.
`request->send_p(200, "text/html", html);`

Even if processor is so simple that:
```
String processor(const String& var) {
  return "";
}

```

nonnullish, 2024-09-25:
I have just run into the exact same issue. I think the problem is that the `Content-Type` and `Content-Length` headers are missing when using a processor. As a work-around, I added the `Content-Type` header, but I'm not sure yet how to calculate the content length there...

```cpp
AsyncWebServerResponse* response = request->beginResponse(LittleFS, "/index.html", String(), false, processor);
response->addHeader("Content-Type", "text/html; charset=UTF-8");
request->send(response);
```

hakkapet, 2024-09-26:
I examined response (in Edge, only have iphone so cannot debug) and library adds Content-Type header, even without addHeader. Response Transfer-Encoding is "chunked" so Content-Length is omitted. https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Transfer-Encoding 
However i havent yet explored how that lib handles chunked responses. I think it is something to do with that.

nonnullish, 2024-09-26:
Apparently there's a PR open with a fix for this issue: https://github.com/me-no-dev/ESPAsyncWebServer/pull/1301/ (linking [this discussion](https://github.com/esphome/ESPAsyncWebServer/issues/42) for more context).

hakkapet, 2024-09-27:
@nonnullish  thanks! i checked that another fork by @mathieucarbou it is much better and works also with IOS18

mathieucarbou, 2024-09-27:
Yes, we are maintaining it at https://oss.carbou.me/ESPAsyncWebServer/ and it has a lot more features, like middlewares ;-)

closed 2024-09-30 by hakkapet