me-no-dev/ESPAsyncWebServer #1437: ESP Async WebServer by Me-No-Dev V3.3.12 incompatibility with Async TCP by Me-No-Dev V3.2.8 and Async UDP by Me-No-Dev V1.1.0.

13Avallejo82, 2024-10-09:
Until last night my ESP32 code was working properly, but today when I opened the Arduino IDE I got the message that there are new versions specifically of the WebServer by Me-No-Dev library, before this update I was working with version 3.0.6 of the ARDUINO library repository, but today version 3.3.12 appeared which has caused my ESP32 code to stop working and see the following error message on the serial monitor:

E (139421) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (139421) task_wdt:  - async_tcp (CPU 0/1)
E (139421) task_wdt: Tasks currently running:
E (139421) task_wdt: CPU 0: IDLE0
E (139421) task_wdt: CPU 1: loopTask
E (139421) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x400E9570:0x3FFC03AC |<-CORRUPTED

E (139421) task_wdt: Aborting.
E (139421) task_wdt: Print CPU 1 backtrace

Backtrace: 0x400822e7:0x3ffb1f90 0x4008214f:0x3ffb1fb0 0x40082157:0x3ffb1fd0 0x40088d29:0x3ffb1ff0 0x40081b71:0x3ffb2010 0x400e7e51:0x3ffb2050 0x400eea5e:0x3ffb2080 0x4017024d:0x3ffb20a0 0x400ef6f1:0x3ffb20c0 0x400ed9ad:0x3ffb2130 0x400edf3a:0x3ffb2180 0x400ee8b5:0x3ffb21f0 0x400ed559:0x3ffb2210 0x400d466e:0x3ffb2240 0x400e3a78:0x3ffb2270 0x4008e796:0x3ffb2290

ELF file SHA256: ba8ca71e2751c555

Rebooting...
ets Jul 29 2019 12:21:46

It seems that a compatibility bug with the Async TCP library appeared unintentionally, even with its latest version V3.2.8, which causes a malfunction in the ESP32 coding, I would simply like to bring it to your attention so that you can analyze it, since I have returned to the version 3.0.6 of ESP Async WebServer and my ESP32 code is working properly again.

Pablo2048, 2024-10-10:
As you can see, this repository has no release with version 3.3.12 so you are in wrong repo. You are probably using this https://github.com/mathieucarbou/ESPAsyncWebServer maintained fork...

PhotoKevin, 2024-10-11:
I ran into the same issue. The odd thing is that the Arduino IDE claims that there is a 3.3.12 version by me-no-dev.
[image]

I can't fathom what's going on, but it's probably going to bite a fair number of people.

13Avallejo82, 2024-10-11:
> As you can see, this repository has no release with version 3.3.12 so you are in wrong repo. You are probably using this https://github.com/mathieucarbou/ESPAsyncWebServer maintained fork...

But, Arduino IDE claims as @PhotoKevin says to upgrade to ESP Async WebServer version 3.3.12, and the title is:

 ESP Async WebServer by Me-No-Dev

[image]

PhotoKevin, 2024-10-11:
I dug about a little more, and it's totally not Me-No-Dev's issue. When I back off to 3.0.6, and look in Documents\Arduino\Libraries\, I find that it's also the mathieucarbou version.

Digging a bit deeper, I see that they're putting this in the library.properties

```
name=ESP Async WebServer
version=3.0.6
author=Me-No-Dev
maintainer=Mathieu Carbou <mathieu.carbou@gmail.com>
sentence=Asynchronous HTTP and WebSocket Server Library for ESP32, ESP8266 and RP2040
paragraph=Supports: WebSocket, SSE, Authentication, Arduino Json 7, File Upload, Static File serving, URL Rewrite, URL Redirect, etc
category=Other
url=https://github.com/mathieucarbou/ESPAsyncWebServer
architectures=*
license=LGPL-3.0
```
Since Arduino only shows the author, I think I'm using code from Me-No-Dev when I'm not. It's nice that they're giving credit, but they really ought to set the author to something like Me-No-Dev/Mathieu Carbou

Pablo2048, 2024-10-12:
Then suggest the changes in "their" repository (if you blame Mathieu), or Arduino for not displaying all informations...

EParisot, 2024-10-12:
Just to mention that the lib's version from ArduinoIDE upgraded to 3.3.13 recently and I can confirm I suffered from it today..., the lib's author is tagged as Me-No-Dev, BUT, the link provided points to Mathieu's repo...

I use ArduinoIDE 2.3.3

Did uninstall, and re-install from this github repo's master manually, but got same issue : 

```
E (37040) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (37040) task_wdt:  - async_tcp (CPU 0/1)
E (37040) task_wdt: Tasks currently running:
E (37040) task_wdt: CPU 0: IDLE0
E (37040) task_wdt: CPU 1: loopTask
E (37040) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x400EA388:0x3FFC044C |<-CORRUPTED
```

mathieucarbou, 2024-10-13:
Copy of my answer at https://github.com/me-no-dev/ESPAsyncWebServer/issues/1437 posted here:

-----

> I certainly see why you wouldn't want to claim ownership of the original code,

That's the opposite. Please read:

- https://forum.arduino.cc/t/ide-library-manager-handling-of-duplicate-library-names/1306491/14
- https://github.com/arduino/library-registry/pull/4585

Note: I've already raised myself some issues regarding how libraries are handled within Arduino Registry.
IMO, it's a mess. They are not handling dependencies correctly. 

For what is worth, and Arduino Registry guys agree with me: **author field should not ne changed, maintainer field has to be added, attribution has to remain.**

**As a reminder, mathieucarbou/ESPAsyncWebServer is a maintained fork by me and some other people**

Also, I would say: stop using toy IDEs like Arduino IDE.  You have real development IDE already available like Vscode and the **pioarduino** extension which provides a good and more reliable dependency management.

The day Arduino IDE will use a correct dependency management system, my opinion will change, but for now, the team behind it does not focus on developers writing code.

mathieucarbou, 2024-10-13:
> Backtrace: 0x400822e7:0x3ffb1f90 0x4008214f:0x3ffb1fb0 0x40082157:0x3ffb1fd0 0x40088d29:0x3ffb1ff0 0x40081b71:0x3ffb2010 0x400e7e51:0x3ffb2050 0x400eea5e:0x3ffb2080 0x4017024d:0x3ffb20a0 [...]

@13Avallejo82 : please de code your stack trace: these traces are useless without your elf file. thanks!

mathieucarbou, 2024-10-13:
> I ran into the same issue.

Which issue ? Can you post a trace please ?

mathieucarbou, 2024-10-13:
To everybody: with some people we are trying to troubleshoot a difficult issue, see:

- https://github.com/mathieucarbou/AsyncTCP/issues/26
- https://github.com/tbnobody/OpenDTU/discussions/2326#discussioncomment-10923684

Could you please rollback to:

- mathieucarbou/AsyncTCP @ 3.2.8
- mathieucarbou/ESPAsyncWebServer @ 3.3.12

And tell me if you still have the issue ?

Thanks!

mathieucarbou, 2024-10-13:
I am referencing this discussion here: https://github.com/tbnobody/OpenDTU/discussions/2326#discussioncomment-10927299 about an issue we are trying to solve.

`mathieucarbou/AsyncTCP @ 3.2.10` is released and might help I think - but needs testing (works on my side).

PhotoKevin, 2024-10-13:
I'm pretty sure that the crash I'm getting when I bump from 3.0.6 to 3.2.10 is a dependency issue and thus my problem to solve. At least I won't bother with crash dumps until I'm sure I'm running the correct code. 

Where I dropped it yesterday was not finding the required TCP library (https://github.com/mathieucarbou/esphome-ESPAsyncTCP/tree/v2.0.0) listed in the Arduino IDE. The only ESPAsychTCP library available is the one from dvarrel. 

(This project is using an ESP8266, changing to ESP32 is not an option)

mathieucarbou, 2024-10-13:
Yes the ESPhome team is not deploying to Arduino registry.
So you need to manually install it or point to another one.
Anyway... These libs didn't change since years and Esp8266 EOL is soon.