me-no-dev/ESPAsyncWebServer #1442: getting undefined reference to `pxCurrentTCB'

joeysmart, 2024-10-23:
when building with esp-idf v5.3.1 for esp32s3 I am getting:
-devkitm-1/lib751/libESP Async WebServer.a(AsyncWebSocket.cpp.o):(.literal._ZN17AsyncWebLockGuardC5ERK12AsyncWebLock[_ZN17AsyncWebLockGuardC5ERK12AsyncWebLock]+0x0): undefined reference to `pxCurrentTCB'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32-s3-devkitm-1/firmware.elf] Error 1

seems pxCurrentTCB is now pxCurrentTCBs and is an array (SMP)
is referenced in the class AsyncWebLock 
currently I have no idea how to fix it

sepp89117, 2025-01-07:
Hi, I just had the same problem.
Replace
```cpp
  bool lock() const {
    extern void *pxCurrentTCB;
    if (_lockedBy != pxCurrentTCB) {
      xSemaphoreTake(_lock, portMAX_DELAY);
      _lockedBy = pxCurrentTCB;
      return true;
    }
    return false;
  }
```
with
```cpp
  bool lock() const {
    TaskHandle_t currentTask = xTaskGetCurrentTaskHandle();
    if (_lockedBy != currentTask) {
        xSemaphoreTake(_lock, portMAX_DELAY);
        _lockedBy = currentTask;
        return true;
    }
    return false;
  }
```
in `AsyncWebSynchronization.h` and it should work.

qinsmoons, 2025-01-09:
> when building with esp-idf v5.3.1 for esp32s3 I am getting: -devkitm-1/lib751/libESP Async 

> Hi, I just had the same problem. Replace
> 
> ```c++
>   bool lock() const {
>     extern void *pxCurrentTCB;
>     if (_lockedBy != pxCurrentTCB) {
>       xSemaphoreTake(_lock, portMAX_DELAY);
>       _lockedBy = pxCurrentTCB;
>       return true;
>     }
>     return false;
>   }
> ```
> 
> with
> 
> ```c++
>   bool lock() const {
>     TaskHandle_t currentTask = xTaskGetCurrentTaskHandle();
>     if (_lockedBy != currentTask) {
>         xSemaphoreTake(_lock, portMAX_DELAY);
>         _lockedBy = currentTask;
>         return true;
>     }
>     return false;
>   }
> ```
> 
> in `AsyncWebSynchronization.h` and it should work.

thanks，it works！