me-no-dev/ESPAsyncWebServer #1451: ESP32 rebooting when I call Server.begin();

DinuraPasan, 2024-12-09:
### Board Settings
I use Arduino IDE 2.3.3 version
[image]

### Code

```
#ifdef ESP8266
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#elif defined(ESP32)
#include <WiFi.h>
#include <ESPmDNS.h>
#else
#error "Board not found"
#endif

#include <ESPAsyncWebServer.h>

AsyncWebServer Server(80);  // Server port is 80

void notFound(AsyncWebServerRequest *request) {
  request->send(404, "text/plain", "Not found");
}

void setup() {
  Serial.begin(115200);
  WiFi.softAP("ESPAsyncWebServer", "dinura123");
  Serial.println(WiFi.softAPIP());

  if (MDNS.begin("ESP")) {  // esp.local/
    Serial.println("MDNS responder started");
  }

  // Loading home page
  Server.on("/", [](AsyncWebServerRequest *request) {
    String message = "Hello World";
    request->send(200, "text/plain", message);
  });

  // Loading page1
  Server.on("/page1", HTTP_GET, [](AsyncWebServerRequest *request) {
    String message = "This is Page 1";
    request->send(200, "text/plain", message);
  });

  Server.onNotFound(notFound);
  Server.begin();
}

void loop() {}
```

### Error Message
```
13:59:26.515 -> 192.168.4.1
13:59:26.515 -> MDNS responder started
13:59:26.515 -> 
13:59:26.515 -> assert failed: tcp_alloc /IDF/components/lwip/lwip/src/core/tcp.c:1851 (Required to lock TCPIP core functionality!)
13:59:26.515 -> 
13:59:26.515 -> 
13:59:26.515 -> Backtrace: 0x4008246d:0x3ffb2000 0x4008c35d:0x3ffb2020 0x4009260a:0x3ffb2040 0x400efc3f:0x3ffb2170 0x400efdb5:0x3ffb2190 0x400da574:0x3ffb21b0 0x400d9786:0x3ffb2200 0x400d2a26:0x3ffb2220 0x400dd95b:0x3ffb2270 0x4008d0a6:0x3ffb2290
13:59:26.556 -> 
13:59:26.556 -> 
13:59:26.556 -> 
13:59:26.556 -> 
13:59:26.556 -> ELF file SHA256: 36dda5900
13:59:26.556 -> 
13:59:26.784 -> Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
13:59:26.784 -> Debug exception reason: BREAK instr 
13:59:26.784 -> Core  1 register dump:
13:59:26.784 -> PC      : 0x400846ec  PS      : 0x00000016  A0      : 0x40080306  A1      : 0x3ffc5e50  
13:59:26.784 -> A2      : 0x3ffb1edc  A3      : 0x00000000  A4      : 0x00100590  A5      : 0x3ffc5e80  
13:59:26.815 -> A6      : 0x00000000  A7      : 0x3ffc5a10  A8      : 0x00000000  A9      : 0x00000000  
13:59:26.815 -> A10     : 0x00000000  A11     : 0x00004000  A12     : 0x3ffb1e60  A13     : 0x70000005  
13:59:26.815 -> A14     : 0xfffbfff0  A15     : 0x3ffc3fb4  SAR     : 0x00000008  EXCCAUSE: 0x00000001  
13:59:26.816 -> EXCVADDR: 0x00000000  LBEG    : 0x400898d0  LEND    : 0x400898e6  LCOUNT  : 0xffffffff  
13:59:26.848 -> 
13:59:26.848 -> 
13:59:26.848 -> Backtrace: 0x400846e9:0x3ffc5e50 0x40080303:0x3ffc5e80 0x65a5a5a2:0xa5a5a5a5 |<-CORRUPTED
13:59:26.848 -> 
13:59:26.848 -> 
13:59:26.848 -> 
13:59:26.848 -> 
13:59:26.848 -> ELF file SHA256: 36dda5900
13:59:26.848 -> 
13:59:26.848 -> Re-entered core dump! Exception happened during core dump!
13:59:26.848 -> Rebooting...
13:59:26.848 -> ets Jul 29 2019 12:21:46
13:59:26.848 -> 
13:59:26.848 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:59:26.848 -> configsip: 0, SPIWP:0xee
13:59:26.848 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:59:26.887 -> mode:DIO, clock div:2
13:59:26.887 -> load:0x3fff0030,len:4688
13:59:26.887 -> load:0x40078000,len:15448
13:59:26.887 -> load:0x40080400,len:4
13:59:26.887 -> ho 8 tail 4 room 4
13:59:26.887 -> load:0x40080404,len:3196
13:59:26.887 -> entry 0x400805a4
13:59:27.295 -> E (430) esp_core_dump_flash: Corf"յ��data check failed:
13:59:27.295 -> Calculated checksum='50a65a55'
13:59:27.295 -> Image checksum='ffffffff'
```

moralesaugusto, 2024-12-17:
Same issue here....

per1234, 2024-12-20:
There is a duplicate report at https://github.com/me-no-dev/ESPAsyncWebServer/issues/1455

In that report, a workaround is described:

> After reinstalling version 3.0.7, the problem was solved.

(3.0.7 is referring to the version of the "**esp32**" boards platform)

So you can use the Arduino IDE Boards Manager to roll back from the incompatible 3.1.0 version of the "**esp32**" boards platform to the last compatible version 3.0.7.

There is also a response from the library author there:

https://github.com/me-no-dev/ESPAsyncWebServer/issues/1455#issuecomment-2552076181

> will be fixed/updated to work with core 3.1.0

mathieucarbou, 2025-01-12:
FYI, please read:

- https://github.com/mathieucarbou/ESPAsyncWebServer?tab=readme-ov-file#dependencies

- https://github.com/espressif/arduino-esp32/issues/10526

If you use the right dependencies this is fixed, until the author merges back the changed from our community maintained fork into his original repos.